RobotLocomotion/drake #1445: Installation error: Can't find lcm files

paulcox, 2015-11-02:
First time installation, using the instructions for the "from source" procedure for a Win7 64-bit system running MATLAB  r2051b 64-bit and MSVC 2015. Have installed prerequisites and added paths and environment variables as necessary.  Getting errors when I run the initial make:

    "C:\work\drake\drake-distro\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
    "C:\work\drake\drake-distro\pod-build\drake.vcxproj" (default target) (10) ->
    (CustomBuild target) ->
      CUSTOMBUILD : ----- warning : bot2-lcmgl not found [C:\work\drake\drake-distro\pod-build\drake.vcxproj]

    "C:\work\drake\drake-distro\pod-build\ALL_BUILD.vcxproj" (default target) (1) ->
    "C:\work\drake\drake-distro\pod-build\drake.vcxproj" (default target) (10) ->
    (CustomBuild target) ->
      c1xx : fatal error C1083: Cannot open source file: 'C:\work\drake\drake-distro\drake\pod-build\lcmgen\lcmtypes\lcmt_body_motion_data.c': No such file or directory [C:\work\drake\drake-distro\drake\pod-build\drake_lcmtypes.vcxproj] [C:\work\drake\drake-distro\pod-build\drake.vcxproj]
      c1xx : fatal error C1083: Cannot open source file: 'C:\work\drake\drake-distro\drake\pod-build\lcmgen\lcmtypes\lcmt_body_wrench_data.c': No such file or directory [C:\work\drake\drake-distro\drake\pod-build\drake_lcmtypes.vcxproj] [C:\work\drake\drake-distro\pod-build\drake.vcxproj]

 full error log here:

https://drive.google.com/file/d/0B7-KdrI3pwWyN1ZWT2sxSmZfOUk/view?usp=sharing 

Thanks,
Paul

paulcox, 2015-11-02:
Opening the directory pointed to in the first error message, I'm guessing an lcm source file has changed name, adding a "drake" prefix?

[image]

Will poke around a bit, but if someone has any hints, I'm all ears.

paulcox, 2015-11-02:
I found opening drake_lcmtypes.vcxproj, replacing the source files from lcmt_xxxx to drake_lcmt_xxxx, makes the project compile, so maybe I'm onto something. Rerunning the command line make seems to recreate the drake_lcmtypes.vcxproj, therefore clobbering my changes and the build fails again. I gotta figure out what generates that project, and I'm guessing it has to do with cmake. Sorry for the stream of consciousness comments, but maybe it'll help a few in the future to see how to debug similar issues.

paulcox, 2015-11-03:
I'm still unable to make and haven't figured out how these projects get generated, so not sure I can fix it on my own. I assume the repo is broken (for new windows users at least) until this addressed. Could someone take a look?

balajincse, 2015-11-03:
hello paulcox, 
    The License file is important one to proceed the installation of 2015b. I have my license and I successfully installed the software. If you buy the MATLAB 2015b the license will be printed in DVD. or Other option is your could get the key through your academic.
once you having the license file you straight way to installing the MATLAB 2015b otherwise you should able to only access the trail version.
The procedure is create an account in MATLAB websit through this link https://in.mathworks.com/mwaccount/register?uri=http://in.mathworks.com/index.html%3Fs_tid%3Dgn_logo. then login to your account the start the math lab it ask you to login again in the software prompt. do it. once successfully connected to you account to the MATLAB. then hit next to select your license if you have the license into your account it will show automatically.. if you have multiple license then it show the multiple license which is associated to your account.  
if you don't have the license with your account, the activation key is required. which mean you buy this product they gives this. 
after successfully completing the license selection. hit next it displays the other product or tools you may need to installing then select your need tools and hit next. then it shows the size of the tools. 
there is two option 
1. if you having iso or DVD then the installation done very quickly.
2. if you using the installer the it displays the each tool as to be download required. give the total file size needs to be download.
Once the download completed it start installing your MATLAB 2015b.

there is one option you could get license through your academic. I done this to getting my license for my MATLAB. I installed successfully.

here is some link you could get clear information about  how to install MATLAB 2015b.
http://in.mathworks.com/help/install/ug/installing-the-license-manager-on-a-server.html

paulcox, 2015-11-03:
Hello, the installation error I'm having is a problem in the drake build system and not related to MATLAB. My MATLAB is installed and working fine. I'm trying to build the sources to drake as per the "source installation" procedure available here: https://github.com/RobotLocomotion/drake/wiki/Installation-and-Quickstart

RussTedrake, 2015-11-20:
Hi Paul.  It definitely looks like a naming issue with our call to lcm-gen.  I'm trying to reproduce it on a Win64 test machine now.

RussTedrake, 2015-11-20:
ah.  i bet i've found it.  we've got this line hidden in the cmake files:
```
macro(get_package_name lcmtype)
  # extract package name from lcm file.
  # creates variable ${_package_name}
  # todo: make this more robust

  execute_process(COMMAND sed -n -e "s/^.*package[[:space:]]*\\(.*\\);[[:space:]]*$/\\1/p" "${lcmtype}" OUTPUT_VARIABLE _package_name OUTPUT_STRIP_TRAILING_WHITESPACE)

endmacro()
```
i thought we had caught all of our uses of sed (and removed it as a dependency).  That could explain why our build machines are fine, but yours is not.  Apologies!  

I'll see about replacing this with cmake logic.  (For a quick fix, you could install sed)

RussTedrake, 2015-11-20:
@psiorx - can you take a look?  the offending lines are in `drake/cmake/lcmtypes.cmake`

paulcox, 2015-11-21:
Yes, installed sed and make ran to the end and I was then able to run some
examples. Thanks,

RussTedrake, 2015-11-21:
Great!  Thanks for confirming.

PR #1463 (2015-11-24): No More Sed
Updates the cmake submodule to include the following change:

Replaced sed call in lcmtypes.cmake with a more robust platform agnostic solution.

Resolves #1445
closed 2015-11-24 by RussTedrake