RobotLocomotion/drake #1446: Drake Visualization for Quadrotor and Strandbeest not working - messages incompatible?

tarquasso, 2015-11-05:
Can not get Quadrotor and Strandbeest examples to visualize:
```
cd(fullfile(getDrakePath,'examples','Quadrotor'));
runLQR();
```
```
cd(fullfile(getDrakePath,'examples','Strandbeest'));
runWithMotor();
```

The drake visualizer opens,  but no robot is displayed within. 
When running:
```
drake-distro/build/bin/bot-spy 
```
it shows that there are drawing messages being send via lcm, but I suspect that those messages are incompatible with the current Drake Visualizer...

On the other hand, the drake visualizer works for example for AtlasWalking and terrainTest, in Atlas Walking, I can actually see Atlas walking slowly forward and in terrainTest, I can see a block fall onto a terrain.

I am running Matlab 2015Ra on OS X Yosemite 10.10.5.

The same problem happens on Brandon's Mac Book.

Hope someone can fix this?

RussTedrake, 2015-11-05:
Just to confirm, the visualizer that opens is the ddapp (director), correct?

tarquasso, 2015-11-05:
it says "Drake Visualizer" on top of the window and the top task bar says "ddConsoleApp".

rdeits, 2015-11-10:
Confirmed. I can reproduce this on my machine.

rdeits, 2015-11-10:
It looks like a bug in the Director. @patmarion is on it.

patmarion, 2015-11-10:
It looks like the bug is exposed only with newer versions of NumPy.  This is the bug:

x = (1,0,0)
x = np.array(x) / np.linalg.norm(x)

Now it must be:

x = (1,0,0)
x = np.array(x, dtype=float) / np.linalg.norm(x)

I'm worried there might be other locations with code like this not covered by tests.  We'll just have to fix them as they are discovered.  In the case of this particular bug report, one of these bad divisions occurred when adding a cylinder to the scene for drawing.  Fixing that now.

commit 7d2f2578 (2015-11-21): update director external sha1.  fixes #1446
commit e6860cf4 (2015-11-21): Merge pull request #1454 from patmarion/update-director-external

update director external sha1.  fixes #1446
closed 2015-11-21 by RussTedrake