RobotLocomotion/drake #1447: precompiled on mac assumes xcode 5

RussTedrake, 2015-11-06:
A student on edX is getting the `__sincos_stret` error (addressed in our drake FAQ) when running on Mac OS X 10.7.5 using  Xcode is 4.6.2 (he can't upgrade to Xcode 5). 
We shouldn't require xcode.  Is there an extra xcode library that we should be bundling with the precompiled distributions?

```
Invalid MEX-file '/Users/richard/Documents/EdX/Underactuated Robotics/drake-v0.9.11-mac/drake/systems/plants/constructModelmex.mexmaci64':
dlopen(/Users/richard/Documents/EdX/Underactuated Robotics/drake-v0.9.11-mac/drake/systems/plants/constructModelmex.mexmaci64, 6): Symbol
not found: ___sincos_stret
Referenced from: /Users/richard/Documents/EdX/Underactuated
Robotics/drake-v0.9.11-mac/drake/systems/plants/../../../build/lib/libLinearMath.2.83.dylib
Expected in: /usr/lib/libSystem.B.dylib
in /Users/richard/Documents/EdX/Underactuated Robotics/drake-v0.9.11-mac/drake/systems/plants/../../../build/lib/libLinearMath.2.83.dylib

Error in RigidBodyManipulator/createMexPointer (line 2139)
obj.mex_model_ptr = constructModelmex(obj);

Error in RigidBodyManipulator/compile (line 846)
model = createMexPointer(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 60)
model = compile(model); % ideally this would happen on entry into any function...

Error in PlanarRigidBodyManipulator/addRobotFromURDF (line 90)
model = addRobotFromURDF@RigidBodyManipulator(model,urdf_filename,xyz,rpy,options);

Error in PlanarRigidBodyManipulator (line 78)
obj = addRobotFromURDF(obj,urdf_filename,zeros(3,1),zeros(3,1),options);
```

tkoolen, 2015-11-06:
From http://stackoverflow.com/a/19039051/2228557
we might need to add `-mmacosx-version-min=10.8` or something similar to our compiler and linker args for the precompiled builds.

patmarion, 2015-11-11:
The issue might be the Mac 10.7 version, not Xcode version.

I think the best way to make things work on older platforms is to run the build on the oldest version of macosx that drake supports.  Adding compile flags might work too, but I'd be worried you can't control the flags of homebrew dependencies (maybe there are not many homebrew lib dependencies for drake core though).

RussTedrake, 2016-05-01:
closing this in anticipation of #1766

closed 2016-05-01 by RussTedrake