RobotLocomotion/drake #1452: Director launch times out too early on some machines

rdeits, 2015-11-10:
On my laptop, running any example using the director causes the director to open, but I still get 

```
attempting to launch the drake director
Error using BotVisualizer (line 48)
Failed to automatically start up a viewer (or to receive the ack, see https://github.com/RobotLocomotion/drake/issues/317)

Error in RigidBodyManipulator/constructVisualizer (line 1696)
            v = feval(type,arg{:});

Error in TimeSteppingRigidBodyManipulator/constructVisualizer (line 1099)
      v = constructVisualizer(obj.manip,varargin{:});

Error in runPassiveDownhill (line 18)
v = p.constructVisualizer();
```

Just running the example again works fine. It looks like we're not waiting long enough for the director to come up.

patmarion, 2015-11-10:
I think there is a mechanism where matlab waits for an lcm heartbeat message from the visualizer.  But this wait is disabled on Mac OSX because matlab LCM message monitor cannot receive lcm messages at all using java 6.  I think there are some old issues tracking that general problem.

RussTedrake, 2015-11-11:
I’ve removed the mac specific logic some time back; we do rely on an ack but have a timeout (which could be increased). 

See https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/BotVisualizer.m  (line 47)

hongkai-dai, 2015-11-13:
I had this same issue on my DRC laptop and my desktop, both running Ubuntu 14.04. I can launch drake-visualizer from terminal, but I got the same error as Robin did when constructing visualizer in MATLAB.

hongkai-dai, 2015-11-13:
Just realized it is not the same as Robin's issue on my machines. It does not work the second time when I run the constructVisualizer code. Also, when I launch drake-visualizer from the terminal, I can see LCM message appeared on "DRAKE_VIEW_STATUS" and "LCM_SELF_TEST" channel. But when I launch the visualizer from MATLAB, such LCM messages do not show up.

RussTedrake, 2016-05-01:
the return value is given, but of course the viewer is started with an & so it runs in the background.  just confirmed that the return value is zero even if the executable is not present.

will set it to 10 seconds, and add a print for drake-visualizer.out

commit 20ce6f3e (2016-05-10): resolves #1452
closed 2016-05-10 by sherm1