RobotLocomotion/drake #1453: Strandbeest demos are broken

rdeits, 2015-11-10:
The Strandbeest model appears to no longer load properly. Attempting to simulate it gives a very long list of infeasible constraints in getInitialState:

```
>> runWithMotor
Warning: RigidBodyManipulator: velocity limits are not supported yet
> In WarningManager/warnOnce (line 21)
  In RigidBodyManipulator/addRobotFromURDF>parseJoint (line 357)
  In RigidBodyManipulator/addRobotFromURDF>parseRobot (line 105)
  In RigidBodyManipulator/addRobotFromURDF (line 55)
  In RigidBodyManipulator (line 75)
  In TimeSteppingRigidBodyManipulator (line 38)
  In runWithMotor (line 8)
Warning:   31 iteration limit reached

> In NonlinearProgram/mapSolverInfo (line 1551)
  In NonlinearProgram/snopt (line 1427)
  In NonlinearProgram/solve (line 980)
  In DynamicalSystem/resolveConstraints (line 317)
  In RigidBodyManipulator/getInitialState (line 347)
  In TimeSteppingRigidBodyManipulator/getInitialState (line 188)
  In FeedbackSystem/getInitialState (line 96)
  In getInitialState
  In sl (line 15)
  In Simulink.BlockDiagram.getInitialState
  In getSimulinkStateStructure (line 13)
  In DynamicalSystem/stateVectorToStructure (line 657)
  In DynamicalSystem/simulate (line 39)
  In runWithMotor (line 32)

infeasible_constraint_name =

    ' non-penetration: body 1 <-> body 19'
    ' non-penetration: body 1 <-> body 25'
    ' non-penetration: body 1 <-> body 31'
    ' non-penetration: body 1 <-> body 37'
    'pair01_leg1_loop_f_g_x'
    'pair01_leg1_loop_f_g_z'
    'pair01_leg1_loop_f_g_ax'
    'pair01_leg1_loop_f_g_az'
    'pair01_leg1_loop_b_c_x'
    'pair01_leg1_loop_b_c_z'
    'pair01_leg1_loop_b_c_ax'
    'pair01_leg1_loop_b_c_az'
    'pair01_leg1_loop_a_c_x'
    'pair01_leg1_loop_a_c_z'
    'pair01_leg1_loop_a_c_ax'
    'pair01_leg1_loop_a_c_az'
    'pair01_leg2_loop_f_g_x'
    'pair01_leg2_loop_f_g_z'
    'pair01_leg2_loop_f_g_ax'
    'pair01_leg2_loop_f_g_az'
    'pair01_leg2_loop_b_c_x'
    'pair01_leg2_loop_b_c_z'
    'pair01_leg2_loop_b_c_ax'
    'pair01_leg2_loop_b_c_az'
    'pair01_leg2_loop_a_c_x'
    'pair01_leg2_loop_a_c_z'
    'pair01_leg2_loop_a_c_ax'
    'pair01_leg2_loop_a_c_az'
    'pair02_leg1_loop_f_g_x'
    'pair02_leg1_loop_f_g_z'
    'pair02_leg1_loop_f_g_ax'
    'pair02_leg1_loop_f_g_az'
    'pair02_leg1_loop_b_c_x'
    'pair02_leg1_loop_b_c_z'
    'pair02_leg1_loop_b_c_ax'
    'pair02_leg1_loop_b_c_az'
    'pair02_leg1_loop_a_c_x'
    'pair02_leg1_loop_a_c_z'
    'pair02_leg1_loop_a_c_ax'
    'pair02_leg1_loop_a_c_az'
    'pair02_leg2_loop_f_g_x'
    'pair02_leg2_loop_f_g_z'
    'pair02_leg2_loop_f_g_ax'
    'pair02_leg2_loop_f_g_az'
    'pair02_leg2_loop_b_c_x'
    'pair02_leg2_loop_b_c_z'
    'pair02_leg2_loop_b_c_ax'
    'pair02_leg2_loop_b_c_az'
    'pair02_leg2_loop_a_c_x'
    'pair02_leg2_loop_a_c_z'
    'pair02_leg2_loop_a_c_ax'
    'pair02_leg2_loop_a_c_az'
    'pair03_leg1_loop_f_g_x'
    'pair03_leg1_loop_f_g_z'
    'pair03_leg1_loop_f_g_ax'
    'pair03_leg1_loop_f_g_az'
    'pair03_leg1_loop_b_c_x'
    'pair03_leg1_loop_b_c_z'
    'pair03_leg1_loop_b_c_ax'
    'pair03_leg1_loop_b_c_az'
    'pair03_leg1_loop_a_c_x'
    'pair03_leg1_loop_a_c_z'
    'pair03_leg1_loop_a_c_ax'
    'pair03_leg1_loop_a_c_az'
    'pair03_leg2_loop_f_g_x'
    'pair03_leg2_loop_f_g_z'
    'pair03_leg2_loop_f_g_ax'
    'pair03_leg2_loop_f_g_az'
    'pair03_leg2_loop_b_c_x'
    'pair03_leg2_loop_b_c_z'
    'pair03_leg2_loop_b_c_ax'
    'pair03_leg2_loop_b_c_az'
    'pair03_leg2_loop_a_c_x'
    'pair03_leg2_loop_a_c_z'
    'pair03_leg2_loop_a_c_ax'
    'pair03_leg2_loop_a_c_az'
    'pair04_leg1_loop_f_g_x'
    'pair04_leg1_loop_f_g_z'
    'pair04_leg1_loop_f_g_ax'
    'pair04_leg1_loop_f_g_az'
    'pair04_leg1_loop_b_c_x'
    'pair04_leg1_loop_b_c_z'
    'pair04_leg1_loop_b_c_ax'
    'pair04_leg1_loop_b_c_az'
    'pair04_leg1_loop_a_c_x'
    'pair04_leg1_loop_a_c_z'
    'pair04_leg1_loop_a_c_ax'
    'pair04_leg1_loop_a_c_az'
    'pair04_leg2_loop_f_g_x'
    'pair04_leg2_loop_f_g_z'
    'pair04_leg2_loop_f_g_ax'
    'pair04_leg2_loop_f_g_az'
    'pair04_leg2_loop_b_c_x'
    'pair04_leg2_loop_b_c_z'
    'pair04_leg2_loop_b_c_ax'
    'pair04_leg2_loop_b_c_az'
    'pair04_leg2_loop_a_c_x'
    'pair04_leg2_loop_a_c_z'
    'pair04_leg2_loop_a_c_ax'
    'pair04_leg2_loop_a_c_az'
    'pair05_leg1_loop_f_g_x'
    'pair05_leg1_loop_f_g_z'
    'pair05_leg1_loop_f_g_ax'
    'pair05_leg1_loop_f_g_az'
    'pair05_leg1_loop_b_c_x'
    'pair05_leg1_loop_b_c_z'
    'pair05_leg1_loop_b_c_ax'
    'pair05_leg1_loop_b_c_az'
    'pair05_leg1_loop_a_c_x'
    'pair05_leg1_loop_a_c_z'
    'pair05_leg1_loop_a_c_ax'
    'pair05_leg1_loop_a_c_az'
    'pair05_leg2_loop_f_g_x'
    'pair05_leg2_loop_f_g_z'
    'pair05_leg2_loop_f_g_ax'
    'pair05_leg2_loop_f_g_az'
    'pair05_leg2_loop_b_c_x'
    'pair05_leg2_loop_b_c_z'
    'pair05_leg2_loop_b_c_ax'
    'pair05_leg2_loop_b_c_az'
    'pair05_leg2_loop_a_c_x'
    'pair05_leg2_loop_a_c_z'
    'pair05_leg2_loop_a_c_ax'
    'pair05_leg2_loop_a_c_az'
    'pair06_leg1_loop_f_g_x'
    'pair06_leg1_loop_f_g_z'
    'pair06_leg1_loop_f_g_ax'
    'pair06_leg1_loop_f_g_az'
    'pair06_leg1_loop_b_c_x'
    'pair06_leg1_loop_b_c_z'
    'pair06_leg1_loop_b_c_ax'
    'pair06_leg1_loop_b_c_az'
    'pair06_leg1_loop_a_c_x'
    'pair06_leg1_loop_a_c_z'
    'pair06_leg1_loop_a_c_ax'
    'pair06_leg1_loop_a_c_az'
    'pair06_leg2_loop_f_g_x'
    'pair06_leg2_loop_f_g_z'
    'pair06_leg2_loop_f_g_ax'
    'pair06_leg2_loop_f_g_az'
    'pair06_leg2_loop_b_c_x'
    'pair06_leg2_loop_b_c_z'
    'pair06_leg2_loop_b_c_ax'
    'pair06_leg2_loop_b_c_az'
    'pair06_leg2_loop_a_c_x'
    'pair06_leg2_loop_a_c_z'
    'pair06_leg2_loop_a_c_ax'
    'pair06_leg2_loop_a_c_az'
```

rdeits, 2015-12-03:
(waiting for the simulation to finish also now results in the robot being lost somewhere offscreen)

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri