RobotLocomotion/drake #1459: SimplePulley demos fail without snopt

cyndiac, 2015-11-20:
At commit 2e5b3fb on Windows; error message says it can't find snopt even though it was checked in the cmake setup; pulleys in SimplePulley and tension fall down to infinity

rdeits, 2015-12-02:
We've reproduced this in Ubuntu 14.04 as well, but I'm still not sure what's causing it.

rdeits, 2015-12-02:
On further inspection, it might not be snopt-related. The example fails on a binary windows install and on a basic ubuntu source install. Possibly a precompiled snopt issue?

cyndiac, 2015-12-02:
Works after installing Gurobi (only tested on Ubuntu machine)

rdeits, 2015-12-03:
That makes sense: I wasn't able to reproduce this on any of my machines, but they all have systemwide gurobi installs that I've accumulated over time.

RussTedrake, 2016-12-23:
won’t fix.  (moving to c++)

closed 2016-12-23 by RussTedrake