RobotLocomotion/drake #1460: testIK is reliably timing out on the build servers

RussTedrake, 2015-11-21:
<matlab_test>strcmp(test_name,'systems/plants/test/testIK') && strcmp(error_id,'timeout')</matlab_test>

c.f. http://drake006.csail.mit.edu/drake/pull-tkoolen-tk-mex-try-call-functions-into-RobotLocomotion-master_20151110-1818-Continuous

RussTedrake, 2016-04-03:
just looked at some history on cdash
https://drake-cdash.csail.mit.edu/testSummary.php?project=1&name=systems/plants/test/testIK&date=2016-04-02

the failure trend graphic suggests that it has been failing, but all of the listed tests say status is "Passed", even on previous dates.  @jamiesnape -- perhaps you can help me understand that one?

i also see that the passing tests are sufficiently far from the timeout threshold that I suspect any timeout we see is actually a crash.

i don't have a good way to add this to the new known issue logic, so it will cause the build servers to complain violently if it keeps happening.

jamiesnape, 2016-04-04:
The table is all recent results and by the graph there were no failures on  April 3.

On a loosely related note, I am not sure the MATLAB tests are all returning the correct exit codes. Also, when a test does not have a dependency it needs to run, it should be skipped rather than pass.

jamiesnape, 2016-04-04:
Although there should be two failures on April 2, so I will look into that.

RussTedrake, 2016-04-04:
Definitely agree on the skip rather than pass.  Am planning to open an issue on that after my PR goes through.   Problem is many dependencies are only known by matlab.

sammy-tri, 2016-05-02:
This test seems to currently be enabled and passing, and completing well within its 600s timeout.  What remains to be done here?

ggould-tri, 2016-05-25:
Following @sammy-tri 's question, I am declaring three weeks enough to age out this issue.

closed 2016-05-25 by ggould-tri