RobotLocomotion/drake #1462: Dynamics simulator used in Lecture 15?

hjjayakrishnan, 2015-11-22:
Towards the end of the lecture, Prof. Russ, shows the dynamic simulation of the Atlas's end effector. Which software was that?

RussTedrake, 2015-11-22:
Everything I've shown is in drake, and/or drake+director

closed 2015-11-22 by RussTedrake

hjjayakrishnan, 2015-11-22:
Few more questions. Sorry if this is the wrong platform,
1. Is Atlas using the Barret hand?
2. Since you mentioned that for underactuated adapting hands like that of Atlas's doesn't necessarily require solving the grasp synthesis problem, do you just grasp the object based on intuition and then perform a force control? I'd love to know more about the strategy you've employed.

RussTedrake, 2015-11-22:
love the questions, but please ask on the piazza forums for Underactuated Robtoics instead.