RobotLocomotion/drake #1466: checkDependency('snopt') doesn't detect our pre-compiled nonlinearprogram

rdeits, 2015-11-23:
checkDependency('snopt') looks for `snopt.m`, which obviously isn't present in our precompiled version of NonlinearProgramSnoptmex. That means that the 6.832 students can't actually use it, because when checkDependency('snopt') fails, we fall back all the way to fmincon. 

The easy fix is to edit snoptEnabled() to look for NonlinearProgramSnoptmex instead of snopt.m, but really we're dealing with two different dependencies: snopt and nonlinearprogramsnopt and we should probably be checking for them separately.

RussTedrake, 2015-11-24:
@hongkai-dai is our expert here.  Will assign this to him for his opinion/fix.

rdeits, 2015-11-30:
Actually, this isn't quite right. I realized that we have a separate check for 'nonlinearprogramsnoptmex' 

I didn't realize that people needed to explicitly do `setSolver('snopt')`. But on at least one machine, it looked like that was failing, even though the student had precompiled snopt on. I'll reopen this if I can reproduce it.

closed 2015-11-30 by rdeits