RobotLocomotion/drake #1471: Clean up our path resolution issues

rdeits, 2015-11-25:
As Drake grows, we'll need a way to keep track of where everything is in the drake and build folders. A few important use cases are:

* in C++, Matlab, Python, etc. we need to be able to, at runtime, look up the location of a URDF file stored somewhere in the drake tree
* in Matlab, we need to be able to look up stored data files (.mat files of fixed points, for example)
* any C++ libraries need to be able to open the other libraries they depend on, even if they're not in build/lib (e.g. the mex files)

There are some complications to this:
* the relative path between `build` and `drake` is only known at configure-time
* we'd really like to allow users to move the drake-distro folder around without breaking anything
* c++ executables might be run from pod-build/bin or from build/bin
* matlab and python scripts might be run from pretty much anywhere

I think the solution to this will involve a few general tools:

* in Cmake, at configure time, we should record the relative path from `drake` to `build` and from `build` to `drake` and add appropriate `#define`s for those values. 
* we should produce functions getDrakePath and getDrakeBuildPath (or similar) in C++, Matlab, and Python
  * one way to do this would be to compile a simple c program that looks up its own path, and install that executable to build/bin. That information, plus the known relative path from build to drake, would be enough to resolve the path at runtime. We can wrap that function in Matlab and Python easily. 

As @patmarion pointed out, it would also be nice for users to be able to override the drake root folder with an environment variable.

liangfok, 2016-12-08:
I'm going to self-assign this since it's a natural follow-up to #4363.

liangfok, 2016-12-08:
> in Cmake, at configure time, we should record the relative path from drake to build and from build to drake and add appropriate #defines for those values.

The problem with this approach is it is incompatible with Drake binary releases (once we have them). To be compatible with binary releases, we cannot rely on any build-time relative path information. Having / using this build-time relative-path-information is super convenient, however, when it works and thus think it can be offered as a default value but allow users to override the path information as described below.

> As @patmarion pointed out, it would also be nice for users to be able to override the drake root folder with an environment variable.

Yes, I agree the use of an environment variable is a viable option. Something like `DRAKE_ROOT`. However, the pitfalls of environment variables are [well known](https://support.cloud.engineyard.com/hc/en-us/articles/205407508-Environment-Variables-and-Why-You-Shouldn-t-Use-Them). In particular, environment variables tend to introduce another source of variability across platforms that can result in difficult-to-resolve bugs.

Perhaps instead of environment variables, we can define a standard `drake_config.yaml` file that defines the root directory relative to which everything is referenced? This same config file could be used to store `package://` and `model://` paths for use by the parsers (see how the ones used by the examples were corralled into a single method in `drake-distro/drake/examples/examples_package_map.cc` in  #4363).

liangfok, 2017-04-12:
I believe this was resolved by #4363, notably the various `PopulateFoo()` methods added to `drake/multibody/package_map.h`.

closed 2017-04-12 by liangfok