RobotLocomotion/drake #1474: RigidBodyManipulator won't compile

braraki, 2015-11-27:
My RigidBodyManipulator broke and I can't figure out why. When I try to make Drake, it gives me this error:

[ 56%] Building CXX object systems/plants/CMakeFiles/drakeRBM.dir/RigidBodyManipulator.cpp.o
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/RigidBodyManipulator.cpp:3:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/util/drakeGeometryUtil.h:8:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/util/drakeGradientUtil.h:6:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/unsupported/Eigen/AutoDiff:32:
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/unsupported/Eigen/src/AutoDiff/AutoDiffScalar.h:629:148: error:
      no member named 'Options' in 'Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const
      Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const
      Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1,
      1, 0, 73, 1> > >, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> > > >'
  ...AutoDiffScalar<Matrix<typename NumTraits<typename DerType::Scalar>::Real,DerType::RowsAtCompileTime,DerType::ColsAtCompileTime, DerType::Options...
                                                                                                                                     ~~~~~~~~~^
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/Eigen/src/Core/MathFunctions.h:192:44: note: in instantiation
      of template class 'Eigen::NumTraits<Eigen::AutoDiffScalar<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const
      Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const
      Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1,
      1, 0, 73, 1> > >, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> > > > > >' requested here
template<typename Scalar, bool IsComplex = NumTraits<Scalar>::IsComplex>
                                           ^
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/Eigen/src/Core/MathFunctions.h:595:10: note: in instantiation
      of default argument for 'conj_impl<Eigen::AutoDiffScalar<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const
      Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const
      Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1,
      1, 0, 73, 1> > >, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> > > > > >' required here
  return EIGEN_MATHFUNC_IMPL(conj, Scalar)::run(x);
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/Eigen/src/Core/MathFunctions.h:54:60: note: expanded from
      macro 'EIGEN_MATHFUNC_IMPL'
#define EIGEN_MATHFUNC_IMPL(func, scalar) Eigen::internal::func##_impl<typename Eigen::internal::global_math_functions_filtering_base<scalar>::type>
                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<scratch space>:86:1: note: expanded from here
conj_impl
^
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/Eigen/src/Geometry/OrthoMethods.h:36:13: note: in
      instantiation of function template specialization
      'Eigen::numext::conj<Eigen::AutoDiffScalar<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const
      Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const
      Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1,
      1, 0, 73, 1> > >, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> >, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, 73, 1> > > > > >' requested here
    numext::conj(lhs.coeff(1) * rhs.coeff(2) - lhs.coeff(2) * rhs.coeff(1)),
            ^
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/RigidBodyManipulator.cpp:1171:84: note: in instantiation of
      function template specialization 'Eigen::MatrixBase<Eigen::Block<Eigen::Block<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0,
      73, 1> >, 6, -1, 0, 6, -1>, 3, -1, false>, 3, 1, true> >::cross<Eigen::Block<Eigen::Matrix<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0,
      73, 1> >, -1, -1, 0, -1, -1>, 3, 1, false> >' requested here
      J.template block<SPACE_DIMENSION, 1>(row_start, *it).noalias() += Jomega_col.cross(point);
                                                                                   ^
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/RigidBodyManipulator.cpp:3:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/util/drakeGeometryUtil.h:8:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/util/drakeGradientUtil.h:6:
In file included from /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/unsupported/Eigen/AutoDiff:32:
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/build/include/eigen3/unsupported/Eigen/src/AutoDiff/AutoDiffScalar.h:629:148: error:
      no member named 'Options' in 'Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, const
      Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const
      Eigen::Matrix<double, -1, 1, 0, -1, 1> >, const Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1,
      1, 0, -1, 1> > >, const Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, -1, 1> >, const
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, -1, 1> > > >'
  ...AutoDiffScalar<Matrix<typename NumTraits<typename DerType::Scalar>::Real,DerType::RowsAtCompileTime,DerType::ColsAtCompileTime, DerType::Options...
                                                                                                                                     ~~~~~~~~~^
[... 68 lines cut ...]
      function for call to 'conj'
    numext::conj(lhs.coeff(1) * rhs.coeff(2) - lhs.coeff(2) * rhs.coeff(1)),
    ^~~~~~~~~~~~
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/RigidBodyManipulator.cpp:627:77: note: in instantiation of
[... cut ...]
      Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, -1, 1, 0, -1, 1> > > > >]
inline EIGEN_MATHFUNC_RETVAL(conj, Scalar) conj(const Scalar& x)
                                           ^
4 errors generated.
make[6]: *** [systems/plants/CMakeFiles/drakeRBM.dir/RigidBodyManipulator.cpp.o] Error 1
make[5]: *** [systems/plants/CMakeFiles/drakeRBM.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[2]: *** [CMakeFiles/drake.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2

RussTedrake, 2015-11-27:
this code definitely did change with a recent pull request.  can you share the compiler info?

braraki, 2015-11-28:
I'm on Mac OS X 10.10.5 (Yosemite). Matlab 2015a, and XCode 7.1.1. That's all the info I can think of giving right now. I did a git pull as well.

braraki, 2015-11-28:
I'm actually getting a new error now when I try to run, say, "runPassive.m" in KneedCompassGait:

Invalid MEX-file
'/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/constructModelmex.mexmaci64':
dlopen(/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/constructModelmex.mexmaci64,
6): Symbol not found:
__Z21createDrakeMexPointerPvRKNSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEiPP11mxArray_tagS8_S8_
  Referenced from:
/Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/constructModelmex.mexmaci64
(which was built for Mac OS X 10.11)
  Expected in:
  /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/pod-build/lib/libdrakeMexUtil.dylib
   in
  /Users/brandonaraki_backup/Documents/MATLAB/drake-source/drake-distro/drake/systems/plants/constructModelmex.mexmaci64
  
Error in RigidBodyManipulator/createMexPointer (line 2139)
        obj.mex_model_ptr = constructModelmex(obj);

Error in RigidBodyManipulator/compile (line 846)
      model = createMexPointer(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 60)
model = compile(model);  % ideally this would happen on entry into any function...

Error in PlanarRigidBodyManipulator/addRobotFromURDF (line 90)
      model = addRobotFromURDF@RigidBodyManipulator(model,urdf_filename,xyz,rpy,options);

Error in PlanarRigidBodyManipulator (line 78)
        obj = addRobotFromURDF(obj,urdf_filename,zeros(3,1),zeros(3,1),options);

Error in runPassive (line 4)
m = PlanarRigidBodyManipulator('KneedCompassGait.urdf');

RussTedrake, 2015-11-28:
That is what you would expect (since the build failed).  I think you should definitely revert to last week’s master revision so that you aren’t stuck.  We’ll fix it asap.

RussTedrake, 2015-11-28:
one more thing to check: did you run make at the drake-distro level (not just the drake level) after doing your update?  the new autodiff code does require an update to Eigen to compile.

tkoolen, 2015-11-28:
> did you run make at the drake-distro level (not just the drake level) after doing your update? the new autodiff code does require an update to Eigen to compile.

This is what I suspect as well.

braraki, 2015-11-29:
I've been doing make at the drake-source level. I tried reverting to an older commit and it's still not compiling...

tkoolen, 2015-11-29:
What is drake-source? If you followed the installation instructions precisely, you should have a drake-distro directory that contains a drake directory. You should be running make in drake-distro. Please stop by office hours tomorrow if you can't get it to work.

braraki, 2015-11-30:
Sorry, I meant drake-distro. I just made another folder and reinstalled/compiled drake and that works now, but I still get this error whenever I try to run something that has a rigidbodymanipulator:

Invalid MEX-file
'/Users/brandonaraki_backup/Documents/MATLAB/drake-newsource/drake-distro/drake/systems/plants/constructModelmex.mexmaci64':
dlopen(/Users/brandonaraki_backup/Documents/MATLAB/drake-newsource/drake-distro/drake/systems/plants/constructModelmex.mexmaci64,
6): Library not loaded: @rpath/librbmMexFunctions.dylib
  Referenced from:
  /Users/brandonaraki_backup/Documents/MATLAB/drake-newsource/drake-distro/drake/systems/plants/constructModelmex.mexmaci64
  Reason: image not found

Error in RigidBodyManipulator/createMexPointer (line 2139)
        obj.mex_model_ptr = constructModelmex(obj);

Error in RigidBodyManipulator/compile (line 846)
      model = createMexPointer(model);

Error in RigidBodyManipulator/addRobotFromURDF (line 60)
model = compile(model);  % ideally this would happen on entry into any function...

Error in RigidBodyManipulator (line 75)
          obj = addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);

Error in LittleDog (line 8)
      obj = obj@RigidBodyManipulator('LittleDog.urdf',options);

braraki, 2015-11-30:
This is what happens when I run make test:

drake-distro brandonaraki_backup$ make test
CMake Error: The current CMakeCache.txt directory pod-build/CMakeCache.txt is different than the directory /Users/brandonaraki_backup/Documents/MATLAB/drake-source-new/drake-distro/pod-build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
make[1]: *** No rule to make target `test'.  Stop.
make: *** [test] Error 2

tarquasso, 2015-12-03:
I had the same errors when running make, I reverted back to commit 11f81bfe0079b3406ceff82086643321d77ead92, now the make works fine. That is the commit before there was change through Twan that affected the eigen pod. 

braraki told me that he just installed from scratch and was then able to get it to compile.

tkoolen, 2015-12-03:
If you want to use the latest, could you try (in drake-distro directory):
```
make clean-eigen
make
```

tkoolen, 2015-12-03:
Just verified with another patient (@diitsgp) that the above works.

@RussTedrake, possibly a bug in our build logic? AUTO_UPDATE_EXTERNALS was turned on for Aditya.

DiitsGp, 2015-12-03:
verified for both my macbook (el capitan, clang) and my linux box (14.04, gcc/g++)

tarquasso, 2015-12-04:
Thanks Twan!
I checked out the latest on Master and ran in drake-distro directory as you suggested:
```
make clean-eigen
make
```
That solved the problem. I am running mac os yosemite.

rdeits, 2015-12-05:
Running 

```
make clean-eigen
make
```

hasn't fixed the issue for me, nor has wiping out the externals/eigen, pod-build/eigen-prefix, and pod-build/download-eigen-prefix folders. I'm not sure what's going on with my system.

tkoolen, 2015-12-05:
Do you have AUTO_UPDATE_EXTERNALS turned on?

tkoolen, 2015-12-05:
Also, could it be that it's finding system eigen?

rdeits, 2015-12-06:
Good point. It looks like it's probably finding the system eigen, which is a slightly different version. I'll try to figure out how to get around that.

tkoolen, 2015-12-14:
Close this?

closed 2015-12-14 by RussTedrake