RobotLocomotion/drake #1477: Two- vs one-return value implementation of DrakeSystem.dynamics

eric-wieser, 2015-11-28:
I've realized that `CascadeSystem` does not implement the `[xdot, xdotdx]` form of `dynamics`. What's the recommended way to check whether the subsystems implement this form, so that it can be forwarded?

eric-wieser, 2015-11-29:
Also, I've seen mention of "AutoDiff" in other issues - will that make the multi-return-value form obsolete?

RussTedrake, 2015-11-29:
The autodiff means we'll be able to provide more and more of the subsystems and gradients from
C++ with a thinner matlab wrapper.  It doesn't make this obsolete.

The answer to your initial question isn't as trivial as i'd like.  The shirt answer is that you can call geval, and it will supply gradients for the subsytem one way or another.  But there is some performance hit there, so we often special case the short-circuit.  The best example might be the constraintEval method in https://github.com/RobotLocomotion/drake/blob/master/drake/solvers/Constraint.m

Awesome that you're contributing here.  Very much appreciated!

eric-wieser, 2015-11-29:
`geval` is numeric, right? vs `generateGradients`, which is symbolic. Unfortunately `generateGradients` didn't deal with nested structs, so I'm having to roll my own symbolic differentiator ([here](https://gist.github.com/eric-wieser/f15cf5be81d9c4f44cd9)). Haven't managed to massage it back over the original API yet.

>  The short answer is that you can call geval, and it will supply gradients for the subsytem one way or another.

Should trajectory optimization be using `geval` then? At present, it seems unable to handle unimplemented gradients in `dynamics`

I'll see if I can get something working in `CascadeSystem` to pass on gradients. Although that means that `output` needs to return gradients too, which I'm not sure I've seen it do.

RussTedrake, 2015-11-29:
geval has a few modes.  TaylorVar is my matlab version of autodiff.  It's slow, but super convenient to get gradients out of the box (which you can improve later with user gradients).  That's been our workflow.

closed 2015-12-27 by RussTedrake