RobotLocomotion/drake #1479: Should DirtranTrajectoryOptimization be deterministic?

eric-wieser, 2015-11-29:
Because I'm finding that sometimes it generates a different trajectory when run multiple times on the same input.

However, I'm also getting occasional matlab crashes, so this could point to another Eigen memory alignment bug.

eric-wieser, 2015-11-29:
(also, does this belong here or on piazza?)

RussTedrake, 2015-11-29:
here is better than piazza.
others can confirm, but I believe that our understanding is that snopt stores hidden state which can make things non-deterministic.

hongkai-dai, 2015-11-29:
The random result could be caused by SNOPT. If you run
checkDependency('snopt'), and it does not print out any error, then SNOPT
is used as the nonlinear solver. After every call of SNOPT, it caches some
result from the current run (such as the active set, basic variables, etc),
and uses those cached results in the next run. This cached result would
make two consecutive calls of SNOPT to generate two different result, even
on the same problem. To avoid such randomness, make sure you call 'make
classes' or 'megaclear' every time after you call
DirtranTrajectoryOptimization. This will erase the cached result.

hongkai-dai, 2015-11-29:
Sorry, it should be 'clear classes' instead of 'make classes'

RussTedrake, 2015-11-29:
thanks @hongkai-dai .  reading your answer it occurred to me that the crashes might be coming from snopt, too.  it's known that if you hit ctrl-c when you're inside the snopt mex, then it will crash on the next run, unless you run something like megaclear.

eric-wieser, 2015-11-29:
I'm still getting very non-deterministic behaviour (after using `megaclear`). Here's one run of my optimization:

[image]

And here's another

[image]

I don't call `rand` anywhere in my code. Interestingly, the blue line has a constraint to stay on a straight line. This seemed to succeed when run on a fresh matlab instance, but evidently no longer does

eric-wieser, 2015-11-29:
To prove I'm not going insane, closing all the figure windows and trying again gave me this (more expected) result:

[image]

tkoolen, 2015-11-30:
Just to be a little more certain that none of the code you're calling is using anything random, could you try
```
dbstop in rand
dbstop in randn
dbstop in randi
dbstop in sprand
dbstop in sprandn
dbstop in randperm
```
and then run your code?

eric-wieser, 2015-12-01:
Thanks, I think that caught it - I wasn't passing `solveTraj(obj,t_init,traj_init)` its last two arguments

tkoolen, 2015-12-01:
You're welcome, good to hear.

tkoolen, 2015-12-03:
Close the issue?

closed 2015-12-03 by RussTedrake