RobotLocomotion/drake #1485: Bug in default prefix generation for `CoordinateFrame` when name contains `:`

vtjeng, 2015-11-30:
When the `name` of a `CoordinateFrame` you are attempting to construct (directly) contains a colon, and (1) there are fewer than 2 characters following the colon, (2) the dimension `dim` of the frame is not 2, you get an error.

For example:

     CoordinateFrame('ab:c', 1)
     CoordinateFrame('ab:cd', 2)

The issue is caused by [these lines](https://github.com/RobotLocomotion/drake/blame/b3a6fb9f8b783a94fa69470744750a6f965e153d/drake/systems/frames/CoordinateFrame.m#L45-L50).

While no `CoordinateFrame

RussTedrake, 2015-12-27:
beyond just documenting it, i would actually even check for the ':' at the entry to the method and throw an error if it's present.

vtjeng, 2015-12-31:
@RussTedrake I'll work on that when I get back on campus, but I'm just curious - what is the parsing based on a ':' meant to do? I'm assuming it's meant to be used for some special case, but I can't see this documented anywhere.

RussTedrake, 2016-01-01:
I’m embarassed to say that I don’t remember (and can’t find it on a quick screen of the code)

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri