RobotLocomotion/drake #1488: Error when attempting to run IRB140 example

reobaird, 2015-12-02:
Steps to reproduce:

1.  Add snopt to matlab path
2.  Add drake to matlab path.
3.  Navigate to drake-distro/drake/examples/IRB140
4.  Run "runPlanning.m" with default arguements

Error:
```
Subscript indices must either be real positive integers or logicals.
Error in snopt (line 136)
F0 = userFG(x);
Error in RigidBodyManipulator/inverseKinBackend (line 224)
    [x_sol,F,info(i)] =
    snopt(x0,xlow(:,i),xupp(:,i),Flow,Fupp,'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);
    Error in RigidBodyManipulator/inverseKin (line 69)
  [q,info,infeasible_constraint]=
  inverseKinBackend(obj,1,[],q_seed,q_nom,varargin{:},ikoptions);
Error in runPlanning (line 49)
[q_end_nom,snopt_info_ik,infeasible_constraint_ik] =
inverseKin(r,q_start_nom,q_start_nom,...
Error in simulWithHand (line 30)
target_xtraj = runPlanning(x0(1:r.getNumPositions), [0.5, 0.0, 0.5]', options); 
```

Other info:

- Installed student mex version of SNOPT 7.5 + matlab .m files to drake-distro/externals/snopt/matlab
- checkDependency('snopt') returns 1

hongkai-dai, 2015-12-02:
This is due to the different versions of SNOPT. Drake supports SNOPT-7.2, and the snopt call is
```
[x,F,inform,xmul,Fmul] = snopt(x,xlow,xupp,Flow,Fupp,userfun,varargin);
```
In SNOPT 7.5, the snopt call is changed to 
```
[x,F,inform,xmul,Fmul] = snopt( x, xlow, xupp, xmul, xstate,...
					 Flow, Fupp, Fmul, Fstate,...
					 userfun, varargin );
```
Notice that `xmul,xstate,Fmul,Fstate` are added as additional input arguments in SNOPT 7.5. 

Could you use SNOPT-7.2 instead?

reobaird, 2015-12-02:
I am not entirely sure how to go about acquiring 7.2.

I got the student version here: http://ccom.ucsd.edu/~optimizers. You have to fill out a form and they reply with a download link to a .mex file. There is only one download uption.

I then downloaded the .m files from this github repo: https://github.com/snopt/snopt-matlab

Would I need to both request an older version mex file + also figure out what point in time to roll back the repo to?

hongkai-dai, 2015-12-02:
If you want to use SNOPT-7.5, you can also change line 224 of inverseKinBackdend.m to
```
[x_sol,F,info(i)] = snopt(x0,xlow(:,i),xupp(:,i),[],[],Flow,Fupp,[],[],'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);
```
Namely, `xstate,xmul,Fstate,Fmul` are all set to `[]`. You might also need to modify the SNOPT call in line 1411 of NonlinearProgram.m, to add the empty arguments there.

reobaird, 2015-12-02:
Thanks! I've got it working now.

I also had to change line 611 to:
```
[w_sol,F,info] = snopt(w0,xlow,xupp,[],[],Flow,Fupp,[],[],'snoptUserfun',0,1,A,iAfun,jAvar,iGfun,jGvar);
```

closed 2015-12-02 by reobaird