RobotLocomotion/drake #1490: gurobi_mex solver produces incorrect answers

rdeits, 2015-12-02:
I was playing around with the Drake QuadraticProgram class, and it appears that the 'gurobi_mex' solver does not produce correct answers. My optimization is:

```
  nv = 10;
  Q = zeros(nv, nv);
  for j = 2:nv
    Q(j,j) = Q(j,j) + 1;
    Q(j-1, j-1) = Q(j-1, j-1) + 1;
    Q(j, j-1) = Q(j, j-1) - 1;
    Q(j-1, j) = Q(j-1, j) - 1;
  end
  Aeq = zeros(2, nv);
  beq = zeros(2, 1);
  Aeq(1, 1) = 1;
  beq(1) = 2;
  Aeq(2, 10) = 1;
  beq(2) = 9;

  prog = QuadraticProgram(Q, zeros(nv, 1), [], [], Aeq, beq)
```

using the `gurobi` solver gives:

```
ans =

   -0.0000
    1.0000
    2.0000
    3.0000
    4.0000
    5.0000
    6.0000
    7.0000
    8.0000
    9.0000
```

which is correct. Using `gurobi_mex` gives:

```
ans =

    9.0000
    8.9999
    8.9999
    8.9998
    8.9997
    8.9997
    8.9997
    8.9996
    8.9996
    8.9996
```

rdeits, 2015-12-04:
I'd suggest just getting rid of gurobiQPmex entirely. We're not using it anywhere except as one possible way to solve `approximateIK`, and it's apparently broken in some hard-to-debug way.

rdeits, 2015-12-04:
(it's also broken for inequality constraints, too)

RussTedrake, 2016-05-01:
best resolution is to add support for gurobi to the optimization problem in c++ and stop calling the other interfaces.

ggould-tri, 2016-05-25:
Assigning to @naveenoid as he will be integrating the Gurobi solver into C++ which will enable us to migrate away from this bug.

naveenoid, 2016-10-27:
Recent work from @hongkai-dai have superseded my work on Gurobi. Im closing this issue for now.

closed 2016-10-27 by naveenoid