RobotLocomotion/drake #1491: ContactImplicitTrajectoryOptimization incorrectly detects number of contacts

rdeits, 2015-12-03:
`ContactImplicitTrajectoryOptimization.setupVariables()` does:

```
obj.nC = obj.plant.getNumContactPairs;
[~,normal,d] = obj.plant.contactConstraints(zeros(obj.plant.getNumPositions,1));
```

But when it actually generates the contact constraints, it uses the active collision options:

(in `addDynamicConstraints()`):

```
[~,~,~,~,~,~,~,mu] = obj.plant.contactConstraints(zeros(nq,1),false,obj.options.active_collision_options);
```

which results in mu being, in general, smaller than `obj.nC`. I believe the fix is to change, in `setupVariables()`:

```
      [~,normal,d] = obj.plant.contactConstraints(zeros(obj.plant.getNumPositions,1), false, obj.options.active_collision_options);
      obj.nC = size(normal, 2);
```

rdeits, 2015-12-03:
(working on building drake master so I can submit a fix)

mposa, 2015-12-03:
Looks good to me.

PR #1492 (2015-12-07): respect active_collision_options when setting up variables
fixes #1491
commit 0acafd0e (2015-12-07): respect active_collision_options when setting up variables

fixes #1491
closed 2015-12-07 by RussTedrake