RobotLocomotion/drake #1497: Issue when adding time-step constraints

sacepi, 2015-12-05:
I was running Trajectory Optimization for my project with independent time steps (options.time_option = 2) and I tried to add a constraint bounding the time steps using traj_opt.addConstraint(BoundingBoxConstraint(lb,ub),traj_opt.h_inds).

Then, while debugging, it seems that the traj_opt.x_lb(traj_opt.h_inds) and traj_opt.x_lb(traj_opt.h_inds) are indeed limited by the desired values when I put in the constraint I mentioned above, but somehow after sometime, these bounding values disappear, and, as a result, the values of $\Delta T$ are not within the desired interval. I could not find yet why it happens. But it seems to be a bug on Drake.

hongkai-dai, 2015-12-05:
When do these bounds disappear? Could you add a break point at the beginning and the end of `solve` function in `NonlinearProgram.m`, to check if traj_opt.x_lb(traj_opt.h_inds) and traj_opt.x_ub(traj_opt.hinds) are the desired value just before and after calling `solve`?

sacepi, 2015-12-06:
I don't know yet exactly when these bounds disappear, but it is before the solve function begin (I had already added that break point). I was printing the value of  traj_opt.x_ub(traj_opt.h_inds) in some points of my code and just after I add the constraint, I could see that it was the desired value of the upper bound, but suddenly it became +infty, before the optimization start. I think this is a bug because I opened the NonLinearProgram.m file and I saw in the code that the the upper bound is supposed to be minimum between the current upper bound and the upper bound added by that constraint, then it that +infty should not appear.

hongkai-dai, 2015-12-06:
When you call addConstraint, you should return the traj_opt like `
traj_opt=traj_opt.addConstraint(...)`, otherwise if you call
`traj_opt.addConstraint` without the return argument, then traj_opt is not
changed. This is MATLAB syntax, it is not a bug.

closed 2015-12-27 by RussTedrake