RobotLocomotion/drake #1498: HybridTrajectoryOptimization addJumpConstraints Indexing Bug

bronadav, 2015-12-05:
Hi there!
There appears to be a bug with HybridTrajectoryOptimization lines 254 and 262.
Original code:
```
 % add guard constraints, 0 = guard(xm,u) for the transition and
        % 0 >= guard(xm,u) for everything else
        for j=1:length(obj.plant.guard{i}),
...
            guard_fun = @(x,u) guard_constraint_fun(obj.plant,obj.plant.guard{i}{j},x,u);
```
This results in an index out of bounds error, because the guard is being indexed not by the type of mode, but by which number mode this is in the sequence of modes. So, for example, if there is a 3-mode sequence {1, 2, 1}, then for the 3rd mode it should be indexed as `obj.plant.guard{1}`, but instead it's being indexed as `obj.plant.guard{3}`, resulting in an out-of-bounds error for a 2-mode system.

The correct lines of code are:
```
for j=1:length(obj.plant.guard{obj.mode_sequence(i)}),
...
guard_fun = @(x,u) guard_constraint_fun(obj.plant,obj.plant.guard{obj.mode_sequence(i)}{j},x,u);
```
At least, they worked for me.

hongkai-dai, 2015-12-05:
I think you are right. @mposa , what do you think?

bronadav, 2015-12-09:
Also, if you decide to incorporate it, similar changes should also be made on lines 222, 226, 227, and 232.

mposa, 2015-12-15:
Yup, I concur that you're right. @bronadav, since it sounds like you've fixed it on your end, are you interested in submitting a pull request? Otherwise, I'll take care of it.

bronadav, 2015-12-17:
Seeing as I'm working from a binary copy of drake (not associated with git), it'd be easier for me if you submitted the PR.
For reference, my full code for the relevant class is reproduced below:
'''
function obj = addJumpConstraints(obj)
      for i=1:obj.M,
        mode_in = obj.plant.modes{obj.mode_sequence(i)};
        
        if i == obj.M,
          transition_ind = -1;
        elseif any(obj.plant.target_mode{obj.mode_sequence(i)} == 0)
          if length(obj.plant.target_mode{obj.mode_sequence(i)}) > 1
            error('Not handling this case right now, better to specify the target modes of all transitions')
          end
          transition_ind = 1;
        else
          transition_ind = find(obj.plant.target_mode{obj.mode_sequence(i)} == obj.mode_sequence(i+1));
        end
        
        if i < obj.M          
          mode_out = obj.plant.modes{obj.mode_sequence(i+1)};
          
          % add transition constraint, xp = transition(xm,u)
          transition_fun = @(xm,u,xp) transition_constraint_fun(obj.plant,obj.plant.transition{obj.mode_sequence(i)}{transition_ind}, ...
            obj.mode_sequence(i),xm,u,xp);
          
          jump_con = FunctionHandleConstraint(zeros(mode_out.getNumStates,1), zeros(mode_out.getNumStates,1), ...
            mode_out.getNumStates + mode_in.getNumStates + mode_in.getNumInputs, transition_fun);
          
          jump_xind{1} = obj.mode_opt{i}.x_inds(:,end) + obj.var_offset(i); % xm
          jump_xind{2} = obj.mode_opt{i}.u_inds(:,end) + obj.var_offset(i); % u
          jump_xind{3} = obj.mode_opt{i+1}.x_inds(:,1) + obj.var_offset(i+1); % xp
          
          obj = obj.addConstraint(jump_con, jump_xind);
        end
        
        % add guard constraints, 0 = guard(xm,u) for the transition and
        % 0 >= guard(xm,u) for everything else
        
        for j=1:length(obj.plant.guard{obj.mode_sequence(i)}),
          for k=1:obj.N(i),
            if k==obj.N(i) && j==transition_ind,
              guard_ub = 0;
            else
              guard_ub = inf;
            end
            
            guard_fun = @(x,u) guard_constraint_fun(obj.plant,obj.plant.guard{obj.mode_sequence(i)}{j},x,u);
            guard_con = FunctionHandleConstraint(0,guard_ub,mode_in.getNumStates + mode_in.getNumInputs, guard_fun);
            guard_xind{1} = obj.mode_opt{i}.x_inds(:,k) + obj.var_offset(i); % xm
            guard_xind{2} = obj.mode_opt{i}.u_inds(:,k) + obj.var_offset(i); % u
            obj = obj.addConstraint(guard_con,guard_xind);
          end
        end
      end
'''

closed 2016-01-23 by RussTedrake