RobotLocomotion/drake #1499: same optimization code works on one computer but not another...

DiitsGp, 2015-12-06:
I'm trying to run bouncingBrickElasticPlanning from my drake fork. It works on my linux computer, but not on my mac. Optimizing with snopt.

avalenzu, 2015-12-07:
Does it crash on the Mac? Or just fail to find a solution?

DiitsGp, 2015-12-07:
It was failing to find a solution...I've reduced the number of knot points and it's fine for now, but I still think it's weird that I got different results with the same code on different machines.

eric-wieser, 2015-12-08:
Is the program deterministic? [this comment](https://github.com/RobotLocomotion/drake/issues/1479#issuecomment-160659403) has a quick way to verify if you think it should be.

RussTedrake, 2015-12-27:
will close this assuming that it can explained with different random seeds on different platforms.  please reopen if you think it's a real issue.

closed 2015-12-27 by RussTedrake