RobotLocomotion/drake #1501: Drake gets confused by gazebo sensor tag

mauricefallon, 2015-12-07:
drake can't handle this gazebo tag from Valkyrie - presumably because it finds the sensor tag:

    <gazebo reference="leftAnkleRoll">
    <sensor name="leftFootSixAxis" type="force_torque">
      <pose>0.0215646 0.0 -0.051054 3.14 0.0 -1.047</pose>
      <sensor_number id="FT001SIM"/>
      <node id="/sensors/leftFootSixAxis"/>
      <api name="TurbodriverForceTorqueSensor_v1"/>
      <always_on>true</always_on>
      <update_rate>1000</update_rate>
      <force_torque>
        <frame>sensor</frame>
      </force_torque>
    </sensor>
    </gazebo>

I'll dig into it when I have a chance.

david-german-tri, 2016-07-25:
@liangfok, You've been working on parsers, so I've reassigned this to you.  Please prioritize however you see fit.

jwnimmer-tri, 2016-11-01:
I can't tell if this is a matlab-only defect.  If so, we should label it as such and kick it back to RLG.

RussTedrake, 2016-11-01:
i believe that it was a matlab defect (but could happen in c++, too).  from memory: the failure is doing a "find first child with tag blah" with the xml parser could find a child of a child, which in this case meant something different.  we should be careful not to do that.  

the matlab urdf parser is effectively deprecated at this point.  safe to close as "will not fix".

closed 2016-11-01 by RussTedrake