RobotLocomotion/drake #1502: SNOPT info 11 fails

braraki, 2015-12-07:
I'm running a ContactImplicitTrajectoryOptimization. I have added two obstacles (walls) into my world. I may have placed them at bad locations, causing a "Warning: 11 infeasible linear constraints" error. However, SNOPT then crashes.
```
> In NonlinearProgram/mapSolverInfo (line 1551)
  In NonlinearProgram/snopt (line 1427)
  In NonlinearProgram/solve (line 980)
  In DirectTrajectoryOptimization/solveTraj (line 188)
  In ContactImplicitTrajectoryOptimization/solveTraj (line 269)
  In runCITO (line 65) 
Warning:   11 infeasible linear constraints
 
> In NonlinearProgram/mapSolverInfo (line 1551)
  In NonlinearProgram/snopt (line 1427)
  In NonlinearProgram/solve (line 980)
  In DirectTrajectoryOptimization/solveTraj (line 188)
  In ContactImplicitTrajectoryOptimization/solveTraj (line 269)
  In runCITO (line 65) 
Undefined function or variable "empty_grad_cin".

Error in NonlinearProgram/snopt (line 1429)
        if(~isempty(empty_grad_cin))

Error in NonlinearProgram/solve (line 980)
            [x,objval,exitflag,infeasible_constraint_name] = snopt(obj,x0);

Error in DirectTrajectoryOptimization/solveTraj (line 188)
      [z,F,info,infeasible_constraint_name] = obj.solve(z0);

Error in ContactImplicitTrajectoryOptimization/solveTraj (line 269)
      [xtraj,utraj,z,F,info] = solveTraj@DirectTrajectoryOptimization(obj,t_init,traj_init);

Error in runCITO (line 65)
  [xtraj,utraj,ltraj,~,z,F,info] = solveTraj(prog,tf,traj_init);
```

I'm not sure if it should be doing this?...
I'm running Drake on Mac OS X 10.10.5, Matlab 2015a.

hongkai-dai, 2015-12-07:
Thanks! That is a bug in the code. I will fix it.

hongkai-dai, 2015-12-07:
@braraki could you apply the patch in #1503 to make sure the error is gone?

braraki, 2015-12-07:
Ok, but it will take a while to run the simulation.

braraki, 2015-12-07:
@hongkai-dai Thanks, it works now!

PR #1503 (2015-12-09): Bug fix in NonlinearProgram
Fix #1502
closed 2015-12-07 by braraki