RobotLocomotion/drake #1509: Operations on trajectories neither check nor output frames

eric-wieser, 2015-12-10:
For instance, any two trajectories of the same dimension can be added together, and will produce a result with an output frame matching neither of the inputs. A couple of proposals:

* Addition (or in general, elementwise operations in the time axis) should fail if the frames do not match
* To make existing cases work, we should distinguish default frames from deliberate frames
* Addition should succeed if the frames do not match, but output the same frame as both inputs if they do match

Thoughts?

RussTedrake, 2015-12-10:
It would be interesting to try the first proposal, and see how many test failures we get.  The second bullet would be difficult to enforce.  The third seems like likely the best option for now.  Again, the new C++ classes will revisit this (and clean it up a bit)

RussTedrake, 2016-12-23:
won’t fix

closed 2016-12-23 by RussTedrake