RobotLocomotion/drake #1510: Linear constraint not working

reobaird, 2015-12-10:
I am trying to limit velocity as follows in a trajectory optimization:
 
```
vmax = [0.52; 0.52; 0.52; 0.52; 0.52; 0.52];
traj_opt = traj_opt.addStateConstraint(LinearConstraint(-vmax,vmax,eye(6)),2:N,8:13);
%state is size 14 with 7 position variables and 7 velocity variables
```
A lot of the time I get Warning:   31 iteration limit reached. Other times, the optimization completes successfully with no errors but upon looking at the trajectory, the constraint is very clearly violated. 
 
I've tried the following simple test-case, and it still doesn't work (constraint is violated). I have a start position constraint, an end position constraint and the velocity constraint. I am only setting constraints on the actuated joints, and I have set high input limits. I have also tried expanding the timespan and number of knot-points. The objectives should be very trivially feasible. My running cost function is also very trivial. Here is the complete test-case code:

```
function [xtraj,utraj,x0] = trajOpt(r,v,options)

  % r contains a RigidBodyManiputor with 7 joints. 6 of the joints are actuated.

    N = 50; % Number of Steps
    T = 5; % Time
    r.setInputLimits(-100,100); %Input
   
    %Max Velocity
    vmax = [0.52; 0.52; 0.52; 0.52; 0.52; 0.52];  % Max Velocity

    %Initial Condition
    x0 = [0; 0; 0; 0; 0; 0; 0;   %x = [joint 1 - 7)
          0; 0; 0; 0; 0; 0; 0;]; %dx
      
    % Set the initial guess for x and u, should be dim(x) by N and dim(u) by N
    x_init = repmat(x0,1,N);
    u_init = randn(6,N);
      
    tspan = linspace(0,T,N);
      
    x2 = [pi/4; pi/4; pi/4; pi/4; pi/4; pi/4]; %x = [joint 1 - 6)
    
    %Create PPTrajectory Object
    traj_init.x = PPTrajectory(foh(tspan,x_init));
    traj_init.u = PPTrajectory(foh(tspan,u_init));

    traj_init.x = traj_init.x.setOutputFrame(r.getStateFrame);
    traj_init.u = traj_init.u.setOutputFrame(r.getInputFrame);
    
    %Create Dircol Trajectory Optimization Object
    traj_opt =  DircolTrajectoryOptimization(r,N,[0.1 T]);
 
    traj_opt = traj_opt.addStateConstraint(ConstantConstraint(x0),1);
    traj_opt = traj_opt.addStateConstraint(LinearConstraint(-vmax,vmax,eye(6)),num2cell(2:N),8:13);
    traj_opt = traj_opt.addRunningCost(@running_cost);
    traj_opt = traj_opt.addStateConstraint(ConstantConstraint(x2),N,1:6);
        
    traj_opt = traj_opt.setSolver('snopt');
    %traj_opt = traj_opt.setSolver('fmincon'); % Also tried fmincon. Just takes a lot longer.
    tic
    [xtraj,utraj,z,F,info] = traj_opt.solveTraj(tspan,traj_init);
    toc

    v.playback(xtraj);
end

function [f,df] = running_cost(dt,x,u)
     f = sum(u.*u,1);
     df = [zeros(1,1+size(x,1)),2*u'];
end
```
 
Am I setting this up wrong? It seems to be successfully constraining on the last timestep (the velocity seems to be at the upper limit on all joints) but nowhere else.

hongkai-dai, 2015-12-10:
What is the value of the velocity at the knot points, in the optimized trajectory? The interpolated spline can violate the velocity constraint in between knot points, due to the interpolation. But they should be within the bounds at the knot point. You can get the value of the state at the knot points by
'''
x_knots=xtraj.eval(xtraj.getBreaks())
'''

hongkai-dai, 2015-12-10:
Is this still a problem?

r-b, 2015-12-10:
Yes it appears to be, I am trying to debug exactly what is going on still and I will update as soon as I can give some more useful information. It appears that the velocity is correct at the knot points but seems to be exceeding the maximum by very high values between knot points.

Counter-intuitively, setting a higher max velocity actually seems to slow down the resulting trajectory.

I've also tried adding constraints to input and joint acceleration.It may also just be a problem with my formulation.

hongkai-dai, 2015-12-10:
You can check the state acceleration at the knot points. For a cubic spline interpolation used in `DircolTrajectoryOptimization`, the velocity in between the know point should follow a quadratic curve. Thus if the acceleration (which follows a linear curve in between the know points) changes a lot at two consecutive knot points, then the velocity can damp wildly in between. This large acceleration at the knot points might due to the large input limits you set.

RussTedrake, 2015-12-27:
I believe we understood this one to be interpolation related (expected behavior, if not ideal).  Please reopen if you disagree.

closed 2015-12-27 by RussTedrake