RobotLocomotion/drake #1511: Drake hangs/crashes when loading URDF for 2nd time

reobaird, 2015-12-10:
Steps to reproduce, using a URDF which references STL files:

```
%Do the following twice.
arm = RigidBodyManipulator();
arm = addRobotFromURDF(arm,'urdf/arm.urdf');
```

The first time, a number of .obj and .obj.mtl files are generated in the path where the .stl meshes are located. The following output is also produced:

```
>> loadArm
Warning: RigidBodyManipulator: velocity limits are not supported yet 
> In WarningManager/warnOnce (line 21)
  In RigidBodyManipulator/addRobotFromURDF>parseJoint (line 357)
  In RigidBodyManipulator/addRobotFromURDF>parseRobot (line 105)
  In RigidBodyManipulator/addRobotFromURDF (line 55)
  In loadArm (line 3) 
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/base_link.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/base_link.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link1.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link1.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link2.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link2.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link3.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link3.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link4.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link4.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link5.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link5.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link6.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
Warning: Mesh /home/roboteam/apps/drake-distro/drake/examples/Redhawk/meshes/link6.obj ignored because it does not have extension .obj (nor can I find a juxtaposed file with a .obj extension)
LCM: Disabling IPV6 support
LCM: TTL set to zero, traffic will not leave localhost.
```

On subsequent runs, matlab crashes (sometimes indefinite hang, sometimes a "error in mex file" message). No output is generated before the crash.

Workaround:

If I delete all .obj and .obj.mtl files that were automatically generated, it reverts to "case 1": lots of warnings and no crash.

RussTedrake, 2015-12-10:
what OS are you using?  are you hitting ctrl-C in the first run?

reobaird, 2015-12-10:
ubuntu 14.04... no ctrl+c on the first run

hongkai-dai, 2015-12-10:
When it crashes, Does MATLAB pop up a window, with a crash report? Could you post the crash report?

RussTedrake, 2015-12-27:
@reobaird -- it would be good to try to reproduce (and resolve) this.  Could you help us reproduce it?  With a minimal example, or by posting your loadArm.m + urdf + mesh files?

RussTedrake, 2016-01-06:
will close this until we know how to reproduce it.

closed 2016-01-06 by RussTedrake