RobotLocomotion/drake #1518: parameterEstimation Error

tarquasso, 2015-12-14:
I am trying to use the built in parameter estimation tool. 

When running 
```
drake-distro/drake/examples/Acrobot/test/paramEstSyntheticData.m
```
to perform a synthetic parameter estimation without noise for the acrobot, it fails and outputs these errors:

```
Elapsed time is 7.972800 seconds.
Warning: Rank deficient, rank = 6, tol =  1.911572e-09. 
> In Manipulator/parameterEstimation (line 121)
  In paramEstSyntheticData (line 19) 
*** Error(1400): bux[4] is too small
Warning: quadprog exitflag = 2147483647 
> In Manipulator/parameterEstimation (line 178)
  In paramEstSyntheticData (line 19) 
Error using sizecheck (line 49)
Wrong size.  Expected [6  1] but got a [0  0] instead.

Error in Point (line 16)
        sizecheck(p,[frame.dim,1]);

Error in Manipulator/parameterEstimation (line 181)
phat = Point(getParamFrame(obj),exp(log_p_est));

Error in paramEstSyntheticData (line 19)
estimated_parameters = parameterEstimation(r,data);
 

t =

 6.0000
```

I am not sure where to start debugging this.

RussTedrake, 2015-12-27:
fwiw - some of these issues have been addressed in #1537

tarquasso, 2016-01-07:
it does now execute after pulling the newest commit from master.

This is the output:

```
>> paramEstSyntheticData
Elapsed time is 22.961756 seconds.

Parameter estimation results:

  Param  	True    	 Initial	Estimated
  -----  	--------	--------	---------
     b1  	    0.07	    0.10	    0.10
     b2  	    0.12	    0.10	    0.10
    lc1  	    0.57	    0.50	    0.50
    lc2  	    1.31	    1.00	    1.00
    Ic1  	    0.09	    0.08	    0.08
    Ic2  	    0.10	    0.33	    0.33
```

I do not think that it is working properly though. Shouldn't the estimated parameters be somewhat close to the true parameters? Instead the estimated parameters are the same as the Initial parameters... I will look more into it.

hanssusilo, 2016-01-08:
What configuration settings are you using to output this?  These should be at the top and be denoted by 'mode', 'model', 'qddmode' and 'method'

tarquasso, 2016-01-15:
```
mode = 'measnoise';
parameterEstimationOptions.model = 'energetic';
qddmode = 'manipul';
parameterEstimationOptions.method = 'nonlinprog';
parameterEstimationOptions.print_result = 'noprint';

```

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri