RobotLocomotion/drake #1520: Clean up solveMexLCP

rdeits, 2015-12-14:
`solveMexLCP` in RigidBodyManipulator.m could really use a cleanup (actually, the entire method probably shouldn't exist). It only exists to set up the 13 positional arguments of solveLCPmex, which include a lot of very redundant data (a `kinsol` *and* H, C, and B, for example), but it creates a ton of overhead in simulation.

Since solveLCPmex isn't used anywhere else anyway, we can probably rewrite its interface to completely replace solveMexLCP. Of course, if this is all going to C++ anyway relatively soon, then it may not be worthwhile.

psiorx, 2015-12-15:
I think the right thing to do here is to have solveMexLCP just be a mexified entry point to the upcoming TSRBM.cpp

rdeits, 2015-12-15:
Yup, sounds good. I'll avoid rewriting that part for now :)

closed 2016-01-23 by RussTedrake