RobotLocomotion/drake #1522: To-Do list for new C++ System class hierarchy

RussTedrake, 2015-12-15:
* [ ] try to implement static (compile-time) tests for isTimeVarying and isDirectFeedthrough via metaprogramming.  if successful, remove them from the template arguments.
* [ ] re-introduce discrete time support to the Systems base class (DiscreteStateVector template argument, update methods, and isDT/isCT variables or methods).
* [ ] get runLCM working again (initial implementation on my dev branch)
* [ ] implement runPendulumDynamics using RigidBodyTree for dynamics (first pass at RigidBodySystem class)
* [ ] implement static checks for isPolynomial, isAffine, isLinear so that I can static_assert gracefully if possible.  Otherwise methods/macros named something like RequirePolynomial, RequireAffine, RequireLinear might help at least give cleaner compiler errors.

RussTedrake, 2015-12-16:
Recommendations: 
- [ ] use namespace method typed on eigen to implement getCoordinateNames(), and replace operator<< in vector concept with getCoordinateNames().  
- [ ] getRandomState should be a method of vector, not of the system
- [ ] investing removing the System class and just have a System concept.  don't template on StateVector, etc, but require that classes modeling this concept implement that typedef.  Or have a barebones Systems class that enforces the concept, but doesn't take vectors as template arguments.
- [ ] is time invariant, is direct feedthrough, etc, is defined implicitly by which methods are implemented by the user (does output take t or not).  see https://github.com/tkoolen/functional/blob/master/src/functional/Traits.h 
- [ ] overloaded system algorithms would require the derived class to explicitly declare something that allows specialization (can't? write isPolynomial and branch on the result, can only through compile error)
- [ ] use namespace methods to define publishLCM etc so that all lcm-related calls stay in e.g. LCM.h
- [ ] use doxygen groups to help users find relevant namespace methods (e.g. in this list: http://drake002.csail.mit.edu/drake/doxygen_cpp/namespacemembers_func.html )

RussTedrake, 2015-12-17:
Recommendation: label isPolynomial, etc (even isSystem or isVector) using traits; as opposed to inheritance.  can still get some of the benefits of inheritance without the pains: c.f. iterator tags: http://www.cs.helsinki.fi/u/tpkarkka/alglib/k06/lectures/iterators.html#tag-dispatching .  many small traits are better than one big trait. (prefer traits to inheritance here because we don't have access to e.g. Eigen types which we want to model the Vector concept)

RussTedrake, 2015-12-21:
I've made a lot of progress in fleshing out the two designs, and it's time to decide.  @tkoolen , @rdeits , @psiorx -- would appreciate your feedback (and anyone else as well).  

Note: I realized along the way that static enforcement of the time-varying and direct-feedthrough properties is probably too limiting.  I think those need to be settable at runtime (e.g. the answers will change depending on which sensors, disturbance models you load).  That reduces the number of template parameters, and motivates the existence of a base System class in either of the approaches below.  

Option A:  Base system class accepts template parameters for the vector types:

Option B: Every class which models the System concept must implement a template alias to define the StateVector, InputVector, and OutputVector.

Check out the differences e.g. for the Pendulum example here:
https://github.com/RussTedrake/drake/commit/1d98f2081ce3f0e542cd98c1da8ce01c59a30924#diff-8c450a2f606fcb002dd732fa6f17f7a4

Option B might lead to less mistakes (because each type is named explicitly instead of entered in some arbitrary order of template parameters), and has the advantage that the aliases are available directly in the Derived class (can refer to StateVector as well as PendulumState).  But it has the disadvantages that it forces more of the template aliasing syntax on the user, and if they get it wrong it will lead to very obscure compiler errors.  For that reason, I'm leaning towards Option A.

Let me know what you think.

psiorx, 2015-12-21:
I think both methods will lead to obscure compile errors if they are gotten wrong.  Option A might even silently succeed if the template parameters are specified in an incorrect order which could lead to hard to find bugs.

The template alias method seems more explicit to me and I like the availability of the types in derived classes.

It looks like LinearSystem as implemented here is some combination of both Option A and Option B.  It takes template parameters for the State and Input vector types and then defines aliases based on those.  Maybe that's the way to go?

RussTedrake, 2015-12-21:
talked to @rdeits, and he likes Option B better, too.  That's two votes for B.  I'm sold.

RussTedrake, 2015-12-27:
this is now mostly done, and the last few will not be forgotten.

closed 2015-12-27 by RussTedrake