RobotLocomotion/drake #1523: Critical design review of build system

RussTedrake, 2015-12-15:
Opening this issue to facilitate a discussion to review our superbuild using CMAKE external projects.  Some major talking points include:
- how we include third party libraries (some of which are public, while others require licenses).  logic flags like AUTO_UPDATE_EXTERNALS, etc.
- how we can include optional examples, including user-contributed examples, without git bloat
- how we organize/install files.  (headers and libs are installed.  matlab files, urdfs + meshes are not.  etc)

RussTedrake, 2015-12-15:
Recommendation: windows install should not require command line interface.  could include make and pkg-config-lite as external projects, and then change the instructions to open the cmake gui, then open visual studio and hit build.

RussTedrake, 2015-12-15:
Recommendation: superbuild options for externals could have multiple strings (e.g. "ON/OFF" --> "YES/NO/IF NECESSARY/SYSTEM").  no throws an error if you're missing a dependency, system doesn't install but also doesn't warn.

RussTedrake, 2015-12-15:
Recommendation:  We admit that PODS isn't great, but there is not a convention on how to lay out a superbuild project.  If PODS is working for us, probably not worth fixing right away.

RussTedrake, 2015-12-15:
Recommendation: fix gurobi user prompt during build.  discussed lots of possible options, but nothing obviously better than prompting in the superbuild and writing a file (or cache entry) that gets used by gurobi.

RussTedrake, 2015-12-15:
Recommendation for avoiding code bloat: add_test or travis CI script which fails if last two(?) git actions add a large file.  idea is to capture the result of a pull request on the build servers.

RussTedrake, 2015-12-15:
Could we install the mex files out of source (even for matlab class methods?  possibly by having two directories with the same class name, one of them in the pod-build directory)?

RussTedrake, 2015-12-15:
recommendation:  change pods_get_base_path.m to have the project name (so that multiple projects do not write the same file)

RussTedrake, 2016-01-12:
- [ ] should support precompiled versions of director (building at least drake-visualizer).  it has many dependencies which bloat the compile time, and many people just need the binary.

rdeits, 2016-02-01:
Re: gurobi credentials: we should be able to use the gurobi.sh shell interface to determine if the user has a working gurobi license, and avoid prompting for a password in that case. That would be extremely helpful for me; I've had to do a lot of clean builds from scratch recently, and having to babysit the build waiting to input my gurobi credentials is very annoying.

RussTedrake, 2016-05-07:
many of these are covered in #1939

closed 2016-05-25 by RussTedrake