RobotLocomotion/drake #1524: Critical design review of documentation strategy

RussTedrake, 2015-12-15:
Our current strategy is
- Use doxygen for inline documentation in matlab and c++
- XML schema (published on build) for documenting URDF file formats
- Primary documentation for examples and tutorials are in course notes: http://underactuated.csail.mit.edu
- Video tutorials (only one or two so far, but people seem to like them)
- User contributed gallery.html 
- External wiki/discussion forum https://github.com/RobotLocomotion/drake/issues/1175
- [Wiki](https://github.com/RobotLocomotion/drake/wiki) for install instructions / FAQ (+ launching point for all of the above)

RussTedrake, 2015-12-15:
Recommendation: consider sphinx to replace wiki documentation (breathe plugin will parse from doxygen output; also a plugin for matlab) so that it is versioned with the source.

RussTedrake, 2015-12-15:
Idea: tutorials in source as iPython notebooks (even with the matlab engine).  can be run as a unit test.

RussTedrake, 2015-12-28:
Also need to figure out a standard for citing publications.  https://github.com/RobotLocomotion/drake/pull/1519#issuecomment-167558130

david-german-tri, 2016-05-25:
@naveenoid, @RussTedrake suggested assigning this to you.  I'll do that once you've added yourself to the TRI group in RobotLocomotion.

tri-ltyyu, 2019-03-22:
@RussTedrake, is this still a concern? if not, please close

EricCousineau-TRI, 2020-06-18:
This still useful? TBH, prolly ain't that hard to write regex or whatever to fix this.

(ofc, this would be queued w/in next few months)

RussTedrake, 2020-06-21:
i'm ok closing it.

closed 2020-06-21 by RussTedrake