RobotLocomotion/drake #1525: Building the drake community

RussTedrake, 2015-12-15:
More contributors potentially means more features and better code (more bug fixes, etc); but we also need to maintain very high standards in terms of code quality and correctness.  Opening this issue to discuss best practices for facilitating use by and contributions from the community.  Ideas welcome.  Here are a few from me:
- Need to provide a very stable and well-documented interface that external code can depend on.  This will happen first in the C++ library interface that we are fleshing out now in preparation for a version 1.0.0 release.
- Need excellent documentation.  see #1524 
- Need contributors to be feel a sense of ownership / pride.  Github gives credit for commits.  Would like people to be able to cite their relevant research papers, etc.  This can currently be accomplished via inline documentation and/or via the [gallery](http://drake002.csail.mit.edu/drake/gallery.html)
- Need better support for ROS workflow.  
- Need to minimize dependencies on expensive software packages / toolboxes.

RussTedrake, 2015-12-15:
Recommendation: github has a convention for CONTRIBUTING.md (or txt, etc) which will put a small line at the top of a pull request page.  "we presume that the contribution is made under the standard license..."

RussTedrake, 2015-12-15:
recommendation: write c++ style guide using clang-format, or uncrustify.  instruct contributors to run this file to format their files before commit.  also editor plugins.

RussTedrake, 2015-12-15:
other ideas: 
- need to figure out the governance model and documenting it. (architecture governance board)
- build from source recommends people take a release tag, instead of master?

mauricefallon, 2015-12-16:
* More information about the pull request review and test process would help: mentioning what Travis is, who will kick off the build servers, that a unit test is usually required with any new capability.
* Summarising notes of developers meetings on the wiki?
* Having a roadmap for future features.

Another developer conference to compare with is [moos-dawg](http://oceanai.mit.edu/moos-dawg)

RussTedrake, 2015-12-16:
Recommendation:  for industry support, compile to real-time OS (RT linux, vxworks, ...)

xuhao1, 2016-01-14:
for RT system, maybe RTAI is a good choice. One of my friend is developing RT-ROS based on this.
I recommend to have a Drake distribution on Docker.

RussTedrake, 2016-01-17:
recommendation: clean up directory structure to capture "modules", as discussed in https://github.com/RobotLocomotion/drake/pull/1612#issuecomment-172355340

closed 2016-05-07 by RussTedrake