RobotLocomotion/drake #1526: Improve build server / continuous integration logic

RussTedrake, 2015-12-15:
We currently have
- Travis CI building with the public externals and running the C++ only unit tests, using g++ and clang on ubuntu.  These spin up fresh virtual machines for every build.
- Custom build scripts / build servers (in my office) running unit tests with all externals and all of the matlab tests, on win32, win64, mac yosemite, ubuntu 12.04 and ubuntu 14.04.  These do not spin up clean machines for every build, and sometimes make implicit assumptions about pre-installed libraries/components, etc.  They also require maintenance.  
We need a better solution.  Soliciting advice.  
- I have toyed with AppVeyor and believe that could serve a role similar to travis for the windows builds.  I haven't yet found a similar solution for Mac.  
- Don't know (yet) how to serve licensed products (e.g., matlab + commercial solvers) on a virtual cloud machine.  But need to find a way.

rdeits, 2015-12-15:
Deploying solver licenses on Travis is actually pretty easy (I have a mosek license in my Iris build server). Travis provides a mechanism for storing an encrypted file in your repo that their server can decrypt at build time.

rdeits, 2015-12-15:
https://docs.travis-ci.com/user/encrypting-files/

rdeits, 2015-12-15:
Correction: the Travis encrypted files aren't available for pull requests from forks (as a security precaution). That hasn't affected IRIS development much, but it might be a big problem for Drake.

RussTedrake, 2015-12-15:
Summary of discussion:
- Need to switch to clean images for every build.  Solutions exist for getting images onto e.g. AWS for ubuntu and windows, but we don't know one for mac.
- Travis will work with mac (but not with matlab).
- The only real reasons we're stuck with custom build server logic is (a) availability of matlab and (b)  known issue logic.  both feel surmountable.  people use Jenkins and BuildBot.

RussTedrake, 2015-12-15:
everyone agrees we want nightly releases, and that we need build servers running tests on those releases (ideally before it was posted, but github has a webhook for release postings that could make it easy to do things serially).

psiorx, 2015-12-15:
> Need to switch to clean images for every build

Maybe we could use Vagrant for something like this?

RussTedrake, 2015-12-16:
Recommendation: turn on coverage tools for C++ now

RussTedrake, 2016-01-08:
great lead from the mathworks: https://github.com/RobotLocomotion/mathworks-requests/issues/11#issuecomment-169677976

rdeits, 2016-01-08:
Very cool!

RussTedrake, 2016-01-12:
Note that we now running appveyor for minimal windows builds.  Need to fill out the build matrix, and enable more externals.  https://github.com/RobotLocomotion/drake/issues/1560

closed 2016-02-27 by RussTedrake