RobotLocomotion/drake #1536: Switch our travis builds to 14.04?

rdeits, 2015-12-16:
Ubuntu 14.04 support was enabled in Travis this fall. Do we want to switch? I'm inclined to say yes, since none of us are running 12.04 anymore.

rdeits, 2015-12-21:
We're going to keep 12.04 on Travis for now. I'll see if it's possible to run both.

rdeits, 2015-12-22:
Things that are too old in the Ubuntu 12.04 package repositories (and that we build from source instead):

* swig (stuck at 2.0)
* python-numpy
* python-scipy
* python-matplotlib

RussTedrake, 2015-12-22:
I see a theme here.  is python going to be the thing that sets our minimum OS?

jhoare, 2015-12-22:
Regarding python, you could consider the conda distribution for older OS's. We use it for older systems where we need a newer python interpreter (e.g. RHEL5/6)

https://www.continuum.io/why-anaconda

rdeits, 2015-12-22:
That's a good point; I hadn't thought about Conda. And no, not yet--building those packages from source is easy, so there's no need to require 14.04

PR #1546 (2015-12-24): add trusty build to travis
closes #1536 
closes #1540
closed 2015-12-24 by RussTedrake