RobotLocomotion/drake #1545: libdrakeIK is incomplete when snopt is not present

rdeits, 2015-12-22:
We still build libdrakeIK when gurobi and snopt are not found, but it results in a library that's totally useless (it only has symbols for a few IKoptions-related functions). 

Should we just disable building that library if snopt and gurobi are missing?

rdeits, 2015-12-22:
(this came up because I spent a few hours trying to figure out why I had linker errors on Travis: it turned out that `libdrakeIK` existed, but just had no symbols in it since I had disabled snopt).

RussTedrake, 2015-12-22:
how hard would it be to enable a useful piece when people have snopt but not gurobi?
maybe we can add it to the snopt precompiled module?

tkoolen, 2015-12-25:
> Should we just disable building that library if snopt and gurobi are missing?
I actually did just that in https://github.com/RobotLocomotion/drake/pull/1514

RussTedrake, 2015-12-25:
ok.  renaming this and making it a feature request

RussTedrake, 2016-05-01:
this is almost resolved with the re-implementation of IK using the new optimization interface.  closing this issue.

closed 2016-05-01 by RussTedrake