RobotLocomotion/drake #1548: SNOPT warning 42: Singular Basis

aespielberg, 2015-12-25:
Hi all,

I hope everyone is having a happy holiday.  I am having the following issue:

I have modified the ContactImplicitTrajectoryOptimization to work with parameters, allowing users to parameterize properties about the robot.  Unfortunately, SNOPT then gives me warning 42: singular basis.  When I add a constraint which says the parameters are fixed, the warning goes away.

So, in short, all I've done is make one of the constants in my f functions a variable, and now I have a singular basis.  In particular, this appears to be an issue with the NonlinearComplementarityConstraint, and only with some parameters.  What I don't get is:

a) How does SNOPT know this is a singular basis?  It's a nonlinear constraint, so what sort of "factorizations" is it doing in order to determine the singularity of the (nonlinear) basis?  What is really meant by a singular basis with SNOPT (I ask because shouldn't it not matter, and in fact be good, if a system of constraints is underconstrained)?
b) Why would making one constant a variable suddenly collapse this to a singular basis?
c) Has anyone had this error in practice?  If so, what does it typically signify, and how did you solve this?

RussTedrake, 2015-12-25:
Hongkai is the resident expert, but errors in the 40s often mean that your user-provided gradients are incorrect.

aespielberg, 2015-12-26:
Hi Russ,

Thanks for the reply.  Based on what I had read in the documentation, error 41 (which I got once and later fixed) is typically caused by bad gradients.  It's possible that other 40 errors are caused by bad constraint gradients, but I believe that 42 is caused bye the evaluation itself.  I tried running my code with all the constraints set to:

constraint.grad_level = 0
constraint.grad_method = 'numerical'

And I had the same error.

Thinking about it more, I'm assuming the factorization is probably something to do with a factorization of the local linearization of the constraints.  I need to think to see if it then would make sense that turning a constant into a variable could make the system singular.

If HongKai has any suggestions or input I'd be interested to know.

hongkai-dai, 2015-12-26:
Hi Andy,

Sorry for my belated reply. I am on a road trip, and just got back online.

What you said is exactly right, info 41 usually means bad gradient, while info 42 usually means bad constraints. My understanding for info 42 is that when solving the QP problem, SNOPT does a QR decomposition of linearized constraint matrix, but that decomposition is singular. My solution for info 42 is usually find out the bad constraint, then modify it.

When you add a constraint to fix that parameter, we (not SNOPT) secretively remove that parameter from your decision variables, and the corresponding column does not appear in the local linearized matrix any more, that is why you see info 42 is gone. You can read line 1273 to 1284 in `NonlinearProgram` for details.

aespielberg, 2015-12-26:
Hi HongKai,

No problem on the delay at all, it was Christmas, I honestly was not even expecting any replies.

I'm mostly confused as to why a singular basis is even bad for the solver.  For instance, let's say I'm trying to solve a constraint:

a + 3 = 0

Obviously, the solution is a = -3.

But now let's say that I change 3 to be a variable:

a + b = 0

Now there's an infinite number of solutions for the constraint.

Shouldn't this be desirable?  Solving the problem just became a lot easier, I would think, as it's now a lot less constrained, and can take an infinite number of values.

My strategy was to create a problem where there's an infinite number of solutions, and hopefully, based on the objective function, the solver picks the "optimal" in that subspace.  Do typical problems in SNOPT rely on only one local feasible point?

hongkai-dai, 2015-12-27:
Hi Andy,

Sorry for the belated reply, I was driving for most of time today.

According to the SNOPT paper `An SQP Algorithm for Large Scale Constrained Optimization`, section 4.3, when solving the QP problem, SNOPT solver needs the constrained matrix to be full row rank; otherwise, the B matrix for its basic variables can be singular. In your problem, could you check the rank of your constrained matrix, when it throws INFO 42? The constraint matrix includes both the nonlinear constraints and linear constraints.

The example you gave in the previous message is for linear constraint. But if you have nonlinear constraint, and you fix one decision variable, then you end up with different optimization problems. These two problems can linearize about different points after several QP calls, and thus the constraint matrices can be quite different. one constraint matrix may have full row rank, while the other does not.

aespielberg, 2015-12-29:
Thanks!  I think I've debugged my problem so I'll close this.  This is good information to know.

closed 2015-12-29 by aespielberg