RobotLocomotion/drake #1561: Issues with Matlab 2012a using precompiled version 0.9.11 (linux)

amcastro0622, 2015-12-29:
I am playing with the examples in the "Underactuated Robotics" notes. For example 1.1  (Robot Manipulators) I issue in the Matlab's command prompt (after changing into the proper directory as in the example):
```matlab
plant = DoublePendPlant;
visualizer = DoublePendVisualizer(plant);
visualizer.inspector(); 
trajectory = simulate(plant,[0 5],randn(4,1));
```
The last command returns the error:

```matlab
Error using DrakeSystem/getModel (line 231) Unknown option specified with the generic
 function of S-function 'DCSFunction' in 'DoublePendPlant_0035033/DrakeSys'. This may be caused by using an S-function generated in a more recent version of Simulink than the one you are currently using. 

Error in DynamicalSystem/simulate (line 38) mdl = getModel(obj);

Error in DrakeSystem/simulate (line 494) [varargout{:}] = simulate@DynamicalSystem(obj,varargin{:});
```

I check my linux and Matlab version issuing a `ver` command in the Matlab prompt. I get:

MATLAB Version: 7.14.0.739 (R2012a)
MATLAB License Number: 837756
Operating System: Linux 3.5.0-17-generic #28-Ubuntu SMP Tue Oct 9 19:31:23 UTC 2012 x86_64
Java Version: Java 1.6.0_17-b04 with Sun Microsystems Inc. Java HotSpot(TM) 64-Bit Server VM mixed mode

MATLAB                                                Version 7.14       (R2012a)
Simulink                                              Version 7.9        (R2012a)
Bioinformatics Toolbox                                Version 4.1        (R2012a)
Communications System Toolbox                         Version 5.2        (R2012a)
Computer Vision System Toolbox                        Version 5.0        (R2012a)
Control System Toolbox                                Version 9.3        (R2012a)
Curve Fitting Toolbox                                 Version 3.2.1      (R2012a)
DSP System Toolbox                                    Version 8.2        (R2012a)
Database Toolbox                                      Version 3.11       (R2012a)
Econometrics Toolbox                                  Version 2.1        (R2012a)
Embedded Coder                                        Version 6.2        (R2012a)
Financial Toolbox                                     Version 4.2        (R2012a)
Fixed-Point Toolbox                                   Version 3.5        (R2012a)
Fuzzy Logic Toolbox                                   Version 2.2.15     (R2012a)
Global Optimization Toolbox                           Version 3.2.1      (R2012a)
Image Acquisition Toolbox                             Version 4.3        (R2012a)
Image Processing Toolbox                              Version 8.0        (R2012a)
Instrument Control Toolbox                            Version 3.1        (R2012a)
MATLAB Builder JA                                     Version 2.2.4      (R2012a)
MATLAB Coder                                          Version 2.2        (R2012a)
MATLAB Compiler                                       Version 4.17       (R2012a)
MATLAB Report Generator                               Version 3.12       (R2012a)
Mapping Toolbox                                       Version 3.5        (R2012a)
Neural Network Toolbox                                Version 7.0.3      (R2012a)
Optimization Toolbox                                  Version 6.2        (R2012a)
Parallel Computing Toolbox                            Version 6.0        (R2012a)
Partial Differential Equation Toolbox                 Version 1.0.20     (R2012a)
RF Toolbox                                            Version 2.10       (R2012a)
Robust Control Toolbox                                Version 4.1        (R2012a)
Signal Processing Toolbox                             Version 6.17       (R2012a)
SimBiology                                            Version 4.1        (R2012a)
SimElectronics                                        Version 2.1        (R2012a)
SimHydraulics                                         Version 1.10.1     (R2012a)
SimMechanics                                          Version 4.0        (R2012a)
SimPowerSystems                                       Version 5.6        (R2012a)
SimRF                                                 Version 3.2        (R2012a)
Simscape                                              Version 3.7        (R2012a)
Simulink 3D Animation                                 Version 6.1        (R2012a)
Simulink Coder                                        Version 8.2        (R2012a)
Simulink Control Design                               Version 3.5        (R2012a)
Simulink Design Optimization                          Version 2.1        (R2012a)
Stateflow                                             Version 7.9        (R2012a)
Statistics Toolbox                                    Version 8.0        (R2012a)
Symbolic Math Toolbox                                 Version 5.8        (R2012a)
System Identification Toolbox                         Version 8.0        (R2012a)
Wavelet Toolbox                                       Version 4.9        (R2012a)

RussTedrake, 2016-01-23:
it looks like you our precompiled distribution on ubuntu is not being built with the sufficiently old version of matlab.  We will try to fix this as we bring up the new batch of build servers.

in the meantime, your best bet is to build from source.

amcastro0622, 2016-01-23:
Yes, that seems the best option for me right now. Thank you.

david-german-tri, 2016-04-25:
Closing this as obsolete.

closed 2016-04-25 by david-german-tri