RobotLocomotion/drake #1562: DirectTrajectoryOptimization fails with collision objects

braraki, 2015-12-30:
I am running drake on Ubuntu 12.0.4.5 and Matlab 2015a.
I am running a DirectTrajectoryOptimization and am trying to include obstacles. I have added collision elements to my urdf and have added obstacles to my 'world.' I get this error when I try to run the optimization:

```
Undefined function 'cos' for input arguments of type 'msspoly'.

Error in rpy2rotmat (line 4)
cos_r = cos(rpy(1));

Error in RigidBodyGeometry.parseURDFNode (line 119)
      T= [rpy2rotmat(rpy),x0; 0 0 0 1];

Error in RigidBodyManipulator/addRobotFromURDF>parseCollision (line 662)
        geometry = RigidBodyGeometry.parseURDFNode(geomnode,xyz,rpy,model,body.robotnum,options);

Error in RigidBodyManipulator/addRobotFromURDF>parseLink (line 189)
    body = parseCollision(body,node.getElementsByTagName('collision').item(collisionItem),model,options);

Error in RigidBodyManipulator/addRobotFromURDF>parseRobot (line 94)
  model = parseLink(model,robotnum,links.item(i),options);

Error in RigidBodyManipulator/addRobotFromURDF (line 54)
model = parseRobot(model,robot,xyz,rpy,options);

Error in RigidBodyManipulator (line 72)
                    obj = addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);

Error in runSwingUpDirtranRobotArm (line 5)
p = RigidBodyManipulator('RobotArmWParamsSmall.urdf');

```

I'm setting up the obstacle like this:

```
p = RigidBodyManipulator('RobotArmWParamsSmall.urdf');
ob1 = RigidBodyBox([1.1,1,3],[0;1;2],[0;0;0]);
ob1.c = [83,53,10]/255;  % brown
p = p.addGeometryToBody('world',ob1, 'wall');
p = compile(p);
```

and setting up the no collision constraint like thisO
```
collision_constraint = generateConstraint(MinDistanceConstraint(p,0.2),0);
prog = prog.addStateConstraint(collision_constraint{1},1:N,1:getNumPositions(p));
```

I've attached the urdf I'm using. I'm wondering if DirectTrajectoryOptimization just won't work with obstacles (in which case I should use a functionhandleconstraint or something like that instead), or if this is a quick fix.

[RobotArmWParamsSmall.urdf.zip](https://github.com/RobotLocomotion/drake/files/74637/RobotArmWParamsSmall.urdf.zip)

RussTedrake, 2015-12-30:
It should work. 

The limitation you’re hitting is that the support for parameters in URDF is only for functions that are polynomial in the parameters.  The good news is, I think the non-polynomial dependence in your URDF is probably just a typo.  I think you meant for the line
```
      <origin rpy="0 0 -$l1 / 2" xyz="0 0 0"/>
```
in the collision element of link1 to be
```
      <origin xyz="0 0 -$l1 / 2" rpy="0 0 0"/>
```

same thing for the collision element of link2.  After those changes, the urdf loads fine for me.

- Russ

braraki, 2015-12-30:
Oh yeah, that was a typo. Thanks and sorry for the trouble!

braraki, 2016-01-04:
I'm getting a similar error again, and this time I don't think it's a typo. I'm getting it when I add a parameter to a force element (a "propellor" on a Quadcopter).

This is the error:
```
Undefined function 'atan2' for input arguments of type 'msspoly'.

Error in rotmat2rpy (line 16)
  rpy = [atan2(R(3,2),R(3,3)); ...

Error in RigidBodyPropellor/addPropellorVisualGeometryToBody (line
167)
      xyz_rpy = [T(1:3,:)*pts; repmat(rotmat2rpy(R*rotmat),1, 1)];

Error in RigidBodyPropellor.parseURDFNode (line 242)
        model = obj.addPropellorVisualGeometryToBody(model, parent,
        visual_diameter);

Error in RigidBodyManipulator/addRobotFromURDF>parseForceElement (line
426)
  [model,fe] =
  RigidBodyPropellor.parseURDFNode(model,name,robotnum,elnode,options);
  
Error in RigidBodyManipulator/addRobotFromURDF>parseRobot (line 115)
  model = parseForceElement(model,robotnum,forces.item(i),options);

Error in RigidBodyManipulator/addRobotFromURDF (line 54)
model = parseRobot(model,robot,xyz,rpy,options);

Error in RigidBodyManipulator (line 72)
                    obj =
                    addRobotFromURDF(obj,filename,zeros(3,1),zeros(3,1),options);
                    
Error in runSwingUpDirtranQuadcopter (line 6)
p = RigidBodyManipulator('Quadcopter.urdf',options);
```

I'm setting the quadcopter up as
```
options.floating = true;
p = RigidBodyManipulator('Quadcopter.urdf',options);
```

I've attached the urdf.

As a note, it ran when I commented out this line:
```
Error in RigidBodyPropellor.parseURDFNode (line 242)
        model = obj.addPropellorVisualGeometryToBody(model, parent,
        visual_diameter);
```

[Quadcopter.urdf.zip](https://github.com/RobotLocomotion/drake/files/76959/Quadcopter.urdf.zip)

RussTedrake, 2016-01-04:
that could be a limitation, but you can at least remove the visual geometry from the propeller and see if you get further.

RussTedrake, 2016-01-06:
did you get any further with this?

braraki, 2016-01-12:
Commenting out the line 242 from RigidBodyPropellor.parseURDFNode
```
        model = obj.addPropellorVisualGeometryToBody(model, parent,visual_diameter);
```
seems to let it work.

closed 2016-01-12 by braraki