RobotLocomotion/drake #1563: Stricter tolerance for FixedPointProgram?

sheim, 2015-12-30:
The tolerances for SNOPT in NonlinearProgram are currently set to 1e-6, but the tilyap() function (called from regionOfAttraction) requires at least 1e-10.  

I think it makes sense to set the tolerances to 1e-10 in the FixedPointProgram (or lower them in tilyap, but 1e-10 doesn't seem too strict?). Should I make a pull-request?

RussTedrake, 2015-12-30:
A pull request would be good, but there is a chance it will break something.  For the most part, my experience in general has been that if snopt can achieve 1e-6 then it can normally achieve 1e-10 almost as quickly.  Thanks.

hongkai-dai, 2015-12-30:
The 1e-6 tolerance in SNOPT is for scaled constraints. It is hard to map this scaled tolerance to the tolerance used in regionOfAttraction.

hongkai-dai, 2015-12-30:
FWIW. The major feasibility tolerance in SNOPT is normalized by the size of the solution. So if the goal is to constrain the violation to be within 1e-10, then the major feasibility tolerance should be set to 1e-10*norm_of_solution.

closed 2016-02-12 by RussTedrake