RobotLocomotion/drake #1569: potentialCollisions returns a comical number of points for a capsule interacting with a flat terrain

RussTedrake, 2016-01-02:
the bullet implementation for multiple_contact support via potentialCollisions works well for e.g. box on box collisions (I think), but seems devastatingly inefficient for more complex geometries.  As an example, simulating a capsule falling into the (very large) box representing flat terrain generates as many points as there are permutations in the potentialCollisions method.  This seems likely to be true for most interesting collision geometries.  
It feels like I cannot turn this on by default.
The question is: is there a better way?  Or is this really the way that people do multi-contact resolution?

avalenzu, 2016-01-07:
> . . . generates as many points as there are permutations in the potentialCollisions method.

So it generates eight contact points? Or am I reading the code wrong?

RussTedrake, 2016-01-08:
I guess that’s probably right.  I thought I remembered seeing many more but am getting 4 contact points in my cylinder simulation now.  (perhaps I saw more when I had more serious penetration issues during my initial debugging).

RussTedrake, 2016-01-11:
i talked to Erwin briefly today, and he says that he does not use this method in the main dynamics pipeline of bullet anymore.  He does some sort of surface clipping algorithm to calculate the polyhedral features.

RussTedrake, 2016-01-11:
also had a great chat w/ Evan Drumright, who has some collision detection code which enumerates all potential collision cases to return the multi-points.  Sounds like it will be super useful to incorporate.

avalenzu, 2016-01-11:
Sounds cool!

closed 2016-01-17 by RussTedrake