RobotLocomotion/drake #1583: Texture map for prius model

RussTedrake, 2016-01-08:


Sproulx, 2016-01-10:
Hopefully I took care of the blocking issues just now. Will get to the rest today/tomorrow.

RussTedrake, 2016-01-10:
i've moved the prius model + simplest stl into drake/examples/Cars in my new pull request.  and updated the alpha for the tires (only) to be 1.0.  might be nice to update the other visual components, too.

RussTedrake, 2016-01-11:
also need to
- [ ] add laser scanners
we have not parsed them from urdf directly before, but do have some urdfs with lidar's specified using gazebo tags:
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Atlas/urdf/atlas_minimal_contact.urdf#L1008
https://github.com/RobotLocomotion/drake/blob/master/drake/examples/PR2/pr2.urdf#L160
it probably makes sense to use (perhaps a minimal form) of that urdf spec.

RussTedrake, 2016-01-11:
my dev branch for all this is at https://github.com/RussTedrake/drake/tree/prius .  it would be great if you could send remaining updates as pull requests to that branch.  you can also compile and run the sim (with carSimLCM path_to_prius_urdf)

also, the scaling on the mesh is close but not perfect, as you suggested.

patmarion, 2016-01-19:
@Sproulx do you have collada or obj files with working textures for the models?  If so, then we can convert those to a format that will load in drake-visualizer.  Let me know what you have (file format) and I'll decide how to proceed.  Basically, drake-visualizer doesn't read collada files, there is a conversion script to write files (with textures) that are ready for consumption by drake-visualizer.

Sproulx, 2016-01-19:
I have both collada and obj files with working textures. Both are just as easy to create on my end, so doesn't matter to me which is used as the starting point.

patmarion, 2016-01-20:
Can you send me the files you have?  I'd like to test the conversion script for compatibility with the files.  If it works then I'll show you how to run the conversion yourself, otherwise I'll try to improve the conversion process.

Sproulx, 2016-01-20:
https://www.dropbox.com/sh/x32g0y4e15qkpuz/AADrTDOyY_v5ktc9BHbdnFvfa?dl=0

Sproulx, 2016-01-20:
That link should bring you to a clone of the directory I'm working from. Both the .dae and .obj contain textures. I can open both in Photoshop and Meshlab and see the textures. Do I ultimately need to have a corresponding .vtp?

patmarion, 2016-01-20:
Here are the new files:

https://www.dropbox.com/sh/yzpirbkcn3l5nyq/AABc777AaXwrw-QLdHo45jJAa?dl=0

You just need to grab the file prius.vtp.  Also included is the script I used called convertOBJ.py.  It reads the .obj file and writes an equivalent .vtp.  I verified that drawing the prius_ackerman.urdf with drake will result in the display of the model with textures.  The urdf file specifies the .dae file, and drake looks for a .obj file, and the visualizer will load the .vtp (or if the vtp file is not available the director loads the .obj file, but without textures).

You can easily run the convertOBJ.py script yourself.  Since you're using Windows, you should use the python program shipped with drake, because that one has access to the VTK python module which is required.  I can show you how to do when I get back next week.  But basically, in the windows cmd prompt, cd to the drake folder and then run:

./build/bin/ddConsoleApp /path/to/convertOBJ.py /path/to/prius.obj

Sproulx, 2016-01-20:
Awesome! Thanks, Pat. I'll definitely be improving the texture, so glad to know this is something I can convert on my own.

closed 2016-01-23 by RussTedrake