RobotLocomotion/drake #1586: passing point clouds to the drake-visualizer

RussTedrake, 2016-01-09:
what is the right way to send point cloud data (e.g. output of the simulated lidar / depth camera) to the drake-visualizer app?  is there a preferred lcm message type?  can we somehow announce the channel name through the drake "bot visualizer" interface?

patmarion, 2016-01-12:
There are several ways to get point clouds into the director visualization environment via LCM messages.  There is support for a generic pointcloud message, drc-maps messages, RGB-D messages (kinect, openni devices), and sensor specific support for multisense and velodyne.  All the methods require C++ dependencies from the openhumanoids stack, for example, lcm message types, C++ conversion routines, and some methods require a PCL dependency.

Typically there is a C++ module in director that receives the pointcloud lcm messages on a thread and calls a conversion routine to extract the lcm message data into a pointcloud data structure, then it converts the pointcloud data structure into a vtk data structure for visualization.  In all existing cases, the lcm messages and conversion routines are provided by the openhumanoids stack, and are not part of the director.  So you only get this functionality when director is compiled in the openhumanoids/main-distro.

For example, last year @mauricefallon added a generic pointcloud lcm message type called drc_pointcloud2_t.lcm which is an lcm duplicate of ROS's sensor_msgs/PointCloud2 message type.  The lcm type and c++ conversion utility live in openhumanoids/main-distro.  Here is the code:

https://github.com/openhumanoids/main-distro/pull/2090/files

The director does not actually define any lcm messages itself, rather it supports messages defined by projects like openhumanoids and drake.  So here are some options we could discuss:

  - add support for drake's existing pointcloud message if one exists
  - add pointcloud message types to drake (copying existing message types from openhumanoids perhaps)
  - add pointcloud lcm messages to director
  - require openhumanoids as a dependency, or extract openhumanoid's lcm messages as a standalone module

One last thing to think about: some message types define whole point clouds, while other message types define single scan lines which are accumulated and stitched together inside director using some additional sensor position messages.  Allowing director to be aware of sensor position and accumulate and fuse scan lines provides maximum flexibility to visualization, but is often sensor specific.

What are your thoughts about introducing additional dependencies to drake to support pointclouds?  PCL is quite popular and has good cross platform support.  It's a very large library, but it provides a good framework for pointcloud development.  There are other options out there, we're also using libpointmatcher.  Extra dependencies are not required, we also have examples converting directly from lcm to vtk data structure for visualization without any extra dependencies.

RussTedrake, 2016-01-12:
I think we should definitely have a point cloud lcm type available in drake.  I think we want to support both the lidar scans (assembled by the viewer) and kinect-like point cloud output.  I will defer to you for which message type is best to bring over.  Adding those types is a small addition and a big gain.  I don't think openhumanoids is a reasonable dependency.

I'd definitely like to have the rendering capability in the drake-visualizer build of director.  How much can director do without PCL?  Does it make sense for director to depend on PCL, but drake to not (at least for now)?  This might matter for people who build drake but grab the precompiled director for just the drake-visualizer.

patmarion, 2016-01-12:
PCL is not mandatory, it's possible for director to provide vtk conversions without it.

My recommendation to get started on this task is to copy over the drc_pointcloud2_t.lcm from openhumanoids and put it in drake, write a drake example that demonstrates drake's sensor simulation and publishes a pointcloud using this message type.  In director, we can add the conversion straight from drc_pointcloud2_t.lcm to vtk for visualization, it's a trivial conversion routine that does not require pcl.  Currently, a conversion routine from openhumanoids is used by director for this lcm message type, but I'm not worried about providing a duplicate implementation because it's very small.  The director already does most of the work, which is the c++ threading and message subscription.

In the future as we add newer features it might make sense to add PCL to the drake superbuild to that it is available to director.  The drake library probably doesn't need pcl unless there is need for non-trivial pointcloud processing in the drake c++ core.

RussTedrake, 2016-01-12:
Sounds good.  Should we change it from drc_pointcloud to the drake relevant name, or is there value in leaving it unchanged?

Agreed on PCL.  Let’s not do it yet, but happy to consider doing it when needed.  Not sure if it should be an external/dep for director (like vtk) or be put directly in the drake superbuild?

mauricefallon, 2016-01-13:
I think there is a need to extend the rendering capabilities to support lines, polygons as well as points. 

Pre-DRC, I worked with some nice tools for doing this (called Collections, originally by Michael Kaess) which did general point cloud and triad rendering (mostly for mapping). I'd like to bring that into the director. 

On the type names:
pointcloud2_t specifically contains a PCL point cloud for simplicity of usage with PCL (and mirrors ROS).

One key design decision ROS folk have made it not to fork messages all the time, rather to define types in independent packages and share them.

mwoehlke-kitware, 2016-02-09:
@patmarion and I talked over this, and aren't convinced that the LCM should necessarily directly encode a PCL data structure. Accordingly, I have created LCM types that use simple data types to convey the interesting data. This makes it easy for consumers to extract the data.

In order to reduce latency, a `lcmt_lidar_data` encodes the data from a single laser pulse (note that "single pulse" may involve multiple independent lasers), and includes the sensor scan angle and direction. This should suffice to cover the three basic types of sensors with which I'm familiar; rotating, bidirectional sweeping, and solid state. (Non-LIDAR data, e.g. CMOS camera with depth estimation, can also be lumped into the last category.) This also gives users the option of consuming the raw stream (minimum latency) or using aggregation to collect a "complete snapshot".

I'll be submitting a PR for director shortly that contains a utility class to perform said aggregation. Currently this class depends on `ddLCMThread` (from director), though it seems like it really ought to live in drake...

david-german-tri, 2016-11-04:
I vaguely recall hearing that someone did this.  @patmarion maybe?

patmarion, 2016-11-04:
drake-visualizer has code to support bot_core.planar_lidar_t and bot_core.pointcloud_t lcm messages.  I'm not aware of any demos in drake that publish these messages, however.

liangfok, 2016-12-22:
I have a WIP branch that's porting System 1.0's `RigidBodyDepthSensor` to System 2.0. I'll go ahead and attempt using `bot_core.planar_lidar_t` and `bot_core.pointcloud_t` to visualize the simulated sensor data.

PR #5779 (2017-04-11): Adds DepthSensorToLcmPointCloudMessage.
Resolves #1586.

Here's a visualization of the first octant point cloud in `drake-visualizer` when the point size is manually set to 5 and colored black:

# First Octant Spec Visualization

```
$ bazel run //drake/systems/sensors:depth_sensor_to_lcm_point_cloud_message_demo
```

[image]

# XY Planar Spec Visualization

```
$ bazel run //drake/systems/sensors:depth_sensor_to_lcm_point_cloud_message_demo -- --spec=xy_planar
```

[image]

# XZ Planar Spec Visualization

```
$ bazel run //drake/systems/sensors:depth_sensor_to_lcm_point_cloud_message_demo --  [... cut]
closed 2017-04-11 by liangfok