RobotLocomotion/drake #1587: simple platform-independent keyboard interface app for steering commands

RussTedrake, 2016-01-09:
I was originally planning to use:
https://github.com/psiorx/steeringdriver

but the keyboard interface runs through linux/input.h . (i'm on a mac, but in general, we should support mac/linux/win).  

This seems like a trivial thing to write in java or python.

crogers3, 2016-01-09:
I'll start writing one up in python.

crogers3, 2016-01-11:
I've finished writing the python version. It requires lcm to be installed on the computer and the pygame python module.

I've put in a pull request at https://github.com/psiorx/steeringdriver, but for now, you can use it from my fork at https://github.com/crogers3/steeringdriver. Running the script with no flags will start in keyboard mode. The window that pops up with the message values must be in focus to publish keyboard commands (but not necessary for joystick).

RussTedrake, 2016-01-12:
it's perfect!  thank you.  i've modified it slightly and added it to my pull request.

closed 2016-01-12 by RussTedrake