RobotLocomotion/drake #1590: need options like "chase camera" in drake-visualizer to support vehicles visualization

RussTedrake, 2016-01-10:
does director support anything like this yet?  if so, can we work it into the BotVisualizer <--> drake-visualizer interface?

peteflorence, 2016-01-11:
:+1: @patmarion did have a proto version of this for me and @manuelli 's class project

patmarion, 2016-01-12:
Currently, drake-visualizer is a thin viewer client.  It doesn't really know what it is drawing.  On the other hand, the Director application used for piloting Atlas is a rich application that knows a lot about the actors in the scene.  We can start adding more features to drake-visualizer inside the application, or add more features to the visualizer lcm message interface so that drake-visualizer remains a thin client.

Here are some design options:

  - provide smart camera logic in drake-visualizer and provide a user interface to it.  The user is required to select the subject of interest and have a user interface to adjust the smart camera behavior.  There could be an lcm interface to the system so that a drake code could initialize the camera behavior automatically.

  - fully define the camera look parameters in lcm messages that are streamed from drake along with the draw messages (which currently contain body transform data).  The drake-visualizer remains a thin client and simply does what it is instructed.  In a log replay, it would reproduce the exact camera motion, but would not be adjustable by the user

Or some combination of these options.  I can imagine the second option is useful for users trying to make a movie and want to fully specify a camera trajectory from their drake code.  The first option lets us implement smart camera behaviors and provide nice user interfaces.

There are some existing camera behaviors in director, but none are exposed via user interface in the drake-visualizer.  Probably the quickest way to get a nice useful camera behavior would be to follow the first option, implement a chase camera in director and add some minimal user interface so the user can click on the subject and enable chase mode.  That allows us to quickly prototype chase camera behaviors to find something we really like, then we can discuss higher design and lcm interfaces.  We could get this working on some of the drake demos.

I have implemented a few tracking camera behaviors in director, but none so far are of the quality you'd get from a 3rd person chase camera found in a typical race car video game.  But it's not hard to get something setup and then we can improve it.

RussTedrake, 2016-01-12:
Sounds awesome.  I agree.  Having an option in director would be the first step.  Perhaps taking a default camera behaviour in the existing load robot message could be an easy second step, that would simply set the director option on load.  more complete camera control over lcm is great, but not needed immediately.

patmarion, 2016-01-26:
Ready for people to try this out.  Several new camera tricks added.  I'll work on getting a pull request, but in the meantime, you can try the branch:

cd externals/director
git remote add patmarion https://github.com/patmarion/director.git
git fetch patmarion
git checkout add-camera-control-panel
make -j

In addition to camera control, this includes some improvements to the drake-visualizer main window (adds a file menu and a view menu to manage dock widgets).

closed 2016-02-05 by RussTedrake