RobotLocomotion/drake #1595: implement default input/output lcm types for urdfLCMNode

RussTedrake, 2016-01-12:
- [ ] EST_ROBOT_STATE for output
  - [ ] take channel name as a command line parameter?  so it can be, e.g. TRUE_ROBOT_STATE instead.
- [ ] ROBOT_COMMAND or similar as the input type

RussTedrake, 2016-05-25:
From @gizatt :

The InstQP needs as input two major things: 
 - EST_ROBOT_STATE, with floating base and joint states populated. 
 - FOOT_CONTACT_ESTIMATE, with binary sensed foot contact states. 
 
drc/software/control/matlab/controllers/LCMBroadcastBlock.m broadcasts these things for the DRC full stack simulation for Atlas (or for Val on a currently-pull-requested branch), alongside some less important hand and Multisense states.

tkoolen, 2016-06-10:
Going to start work on this today.

tkoolen, 2016-06-10:
Related pull request for reference: https://github.com/openhumanoids/oh-distro-private/pull/2383

tkoolen, 2016-09-08:
Updated title.

The current plan for Valkyrie is to change InstantaneousQPController to output torques directly in a new message type, as opposed to the Atlas-specific atlas_command_t. This is only for Valkyrie, not for Atlas. That new message type will be what the simulator will subscribe to.

The only output type coming from the simulator will be robot_state_t, as drc-contact-monitor uses that to determine foot contact state and send a foot_contact_estimate_t on the real robot. We'd like to have a simulator allowing us to test all of the processes running on the real robot without running the real robot, so the simulator should act as a replacement for LCM2ROSControl (minus the conversion from atlas_command_t to torques, as noted earlier) and JointStatePublisher.

RussTedrake, 2016-11-23:
I believe this has been resolved.  @tkoolen -- correct me if I'm wrong.

closed 2016-11-23 by RussTedrake