RobotLocomotion/drake #1598: need a "flatter" method for combining systems

RussTedrake, 2016-01-12:
...more like how simulink stores a .mdl

the nested feedback / cascade combinations is conceptually clean, but has a bad flaw:  the output methods can need to get called lots of times.  For instance, system 1 in a feedback system will have it's output method called twice per DT in the simple case, but if it's nested in another feedback loop, this could be 4x, ...    that's potentially extremely bad because system1 is often our rigid body plant, and might be calling collision detection to simulate lidar, etc.

the alternative is to call the output method for every system in the mdl once, remember the outputs, and apply them as needed for evaluating each dynamics. 

architecting the combination class could also make mimofeedback connection easier.  they are currently not implemented in c++ because it's tough.

have to think through how this could play out with the templated system types.

tkoolen, 2016-01-28:
Just to document this, another option could be to change the separate dynamics + output methods into a single method that returns a pair of state and output.

RussTedrake, 2016-02-19:
Having a good discussion with @rdeits.  

Observation:  PlanEval is almost a system, but has the additional complexity of needing to trigger a discrete change in the state on an event (receiving the plan message).  

Potential solution: Every system could implement (some number of?) event handlers in addition to dynamics, update, and output?

Observation:  it’s really hard to work with rigid body systems built up from multiple robots, due to the multi coordinate frame wiring issues.  

Potential solution: addRobotFromURDF could return a reference to the robot’s inputs/state/output signals so that it could be referred to in later combinations.

david-german-tri, 2016-04-18:
A design doc to fix this (and assorted other limitations of System.h) is here: https://docs.google.com/a/tri.global/document/d/1YQIh2LUETE4CrQ6NvVeNefiIgXch5Q9JO0OWPyMOe90/pub

Email me to request comment access.

david-german-tri, 2016-09-08:
This issue is superseded by System2.  Closing.

closed 2016-09-08 by david-german-tri