RobotLocomotion/drake #1599: runAtlasJumping fails (immediately)

RussTedrake, 2016-01-13:
turned a warning into an error and caught a few bugs.  I suspect this has been failing for a long time.  the first call to solveLCP has u coming in with all nans.  

<matlab_test>strcmp(test_name,'examples/Atlas/runAtlasJumping')</matlab_test>

hongkai-dai, 2016-01-13:
I turned `use_mex` to `0` in QPController, and then the controller does not output `nan`. I will look into the   `QPControllermex` function

rdeits, 2016-02-02:
I'm going to delete the entire QPControllermex implementation and flip this test to use the QPController matlab implementation. QPController is the old controller anyway, and it's only being used for this test and for `runAtlasRunning`, so there's no reason to keep or support its non-functional C++ implementation. 

Eventually, we'll port the running and jumping examples to the new controller anyway.

PR #1683 (2016-02-27): Delete old mex controller, switch jumping and running to matlab controller
This is waiting on #1678 

The old qpcontrollermex implementation is broken and no longer used anywhere, so we should remove it. The atlas jumping example needs to be ported to the new controller anyway, and the running example has been broken since atlas v5. 

Fixes #1599
closed 2016-02-27 by RussTedrake