RobotLocomotion/drake #1600: Thrust failed when using quaternion

xuhao1, 2016-01-13:
I got the same problem when using thrust like https://github.com/RobotLocomotion/drake/issues/1132 and my code is up-to-date.
```matlab
 opts.floating = 'quat';
plant = RigidBodyManipulator(filepath,opts);
plant.dynamics(0,zeros(17,1),zeros(2,1))
```
and got 
```
Subscripted assignment dimension mismatch.

Error in RigidBodyThrust/computeSpatialForce (line 53)
      B_mod(:,obj.input_num) = obj.scale_factor*J'*axis_world;

Error in RigidBodyManipulator/manipulatorDynamics>computeExternalForcesAndInputMatrix (line 87)
        [force,B_force] = computeSpatialForce(obj.force{i},obj,q,v);

Error in RigidBodyManipulator/manipulatorDynamics (line 25)
  [f_ext, B] = computeExternalForcesAndInputMatrix(obj, q, v);

Error in Manipulator/dynamics (line 78)
        [H,C,B] = manipulatorDynamics(obj,q,v);
```
And the state of the plant is
```matlab session
>> plant.getStateFrame
Multi-Coordinate Frame: EternityPosition+EternityVelocity (17 elements)
  EternityPosition (9 elements)
  EternityVelocity (8 elements)
```

RussTedrake, 2016-01-13:
i've pushed a pull request with the fix (i think), but I didn't have a failing test.  can you please try it and let us know?

xuhao1, 2016-01-13:
Hmmm in fact, before you push this commit I have try exactly the same way (follow the propoller code) ,and it will cause error on reshape function
```
Error using reshape
To RESHAPE the number of elements must not change.
```
on line
```
dB_mod((obj.input_num-1)*nq + (1:nq),1:nq) = obj.scale_factor*(J'*daxis_world + reshape(dJ'*axis_world,nq,nq));
```
the cols of dJ is not a square number

xuhao1, 2016-01-13:
Yes, it works.....But the bad news is that will cause problem on file https://github.com/RobotLocomotion/drake/blob/master/drake/systems/plants/@Manipulator/Manipulator.m#L82
```
Matrix dimensions must agree.

Error in Manipulator/dynamics (line 83)
        tau = tau + computeConstraintForce(obj,q,v,H,tau,Hinv);
```

RussTedrake, 2016-01-13:
Try my latest push?

xuhao1, 2016-01-13:
Cool, it works! Thanks a lot!

RussTedrake, 2016-01-13:
If you have an example that we could add (even to a test folder) it would be great to have coverage for this.  Please consider sending a pull request.

xuhao1, 2016-01-13:
OK, I will push a complete VTOL UAV example after when I have made more progress~

xuhao1, 2016-01-13:
Thanks for your job again : )

closed 2016-01-13 by RussTedrake

xuhao1, 2016-01-13:
Hmm...it failed again when I try to make more test
```matlab
opts.floating = 'quat';
plant = RigidBodyManipulator('eternity.urdf',opts);
q_hover = [0;0;0;0.8;0.6;0;0;0;0];
qd = zeros(8,1);
[f0,df0] = plant.dynamics(0,[q_hover;qd],[17.2;17.2])
```
Error message is
```matlab session
Error using reshape
To RESHAPE the number of elements must not change.

Error in RigidBodyThrust/computeSpatialForce (line 46)
        dB_mod((obj.input_num-1)*nv + (1:nv),1:nq) = obj.scale_factor*(J'*daxis_world + reshape(dJ'*axis_world,nv,nv));

Error in RigidBodyManipulator/manipulatorDynamics>computeExternalForcesAndInputMatrix (line 84)
        [force,B_force,dforce,dB_force] = computeSpatialForce(obj.force{i},obj,q,v);

Error in RigidBodyManipulator/manipulatorDynamics (line 23)
  [f_ext, B, df_ext, dB] = computeExternalForcesAndInputMatrix(obj, q, v);

Error in Manipulator/dynamics (line 56)
        [H,C,B,dH,dC,dB] = obj.manipulatorDynamics(q,v);

Error in fucktest (line 5)
[f0,df0] = plant.dynamics(0,[q_hover;qd],[17.2;17.2])
```
seems dim of J makes this error.

```xml
<robot xmlns="http://drake.mit.edu"
	xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
	xsi:schemaLocation="http://drake.mit.edu ../../doc/drakeURDF.xsd" name="Eternity">
	<link
		name="fuselage">
		<inertial>
			<origin
				xyz="0 0 0"
				rpy="0 0 0" />
			<mass
				value="3.5" />
			<inertia
				ixx="0.15"
				ixy="0"
				ixz="0"
				iyy="0.08"
				iyz="0"
				izz="0.15" />
		</inertial>
		<visual>
			<origin
				xyz="0 0 -1.1"
				rpy="0 0 -1.57" />
			<geometry>
				<mesh filename="simple_flyingwing_stl/00_min.OBJ" scale="5"/>
			</geometry>
			<geometry>
				<cylinder length="1" radius="0.0075"/>
			</geometry>
			<material
				name="">
				<color rgba="0.35 0.25 0.28 1"/>
			</material>
		</visual>
		<collision>
			<origin
				xyz="0 0 0"
				rpy="0 0 0" />
			<geometry>
				<!-- <mesh filename="meshes/GliderFuselage.STL" /> -->
				<box size ="1.5 1 .3"/>
			</geometry>
		</collision>
	</link>
	<link name="thrust_nozzle_left">
		<inertial>
		<origin xyz="-0.3 0.12 0" rpy="0 0 0"/>
			<mass value="0.01"/>
			<inertia 
				ixx="0.001"
				ixy="0"
				ixz="0"
				iyy="0.001"
				iyz="0"
				izz="0.001" 
				/>
		</inertial>
		<visual>
			<origin xyz="0 0 0" rpy="0 1.57 0" />
			<geometry>
				<cylinder length="0.1" radius="0.045"/>
			</geometry>
			<material name="">
				<color rgba = "0 1 0 1"/>
			</material>
		</visual>
	</link>
	<link name="thrust_nozzle_right">
		<inertial>
		<origin xyz="-0.3 -0.12 0" rpy="0 0 0"/>
			<mass value="0.01"/>
			<inertia 
				ixx="0.001"
				ixy="0"
				ixz="0"
				iyy="0.001"
				iyz="0"
				izz="0.001" 
				/>
		</inertial>
		<visual>
			<origin xyz="0 0 0" rpy="0 1.57 0" />
			<geometry>
				<cylinder length="0.1" radius="0.045"/>
			</geometry>
			<material name="">
				<color rgba = "0 0 1 1"/>
			</material>
		</visual>
	</link>
	<joint 
		name="nozzle_left"
		type="revolute">
		<origin xyz="-0.3 0.12 0" rpy="0 0 0"/>
		<parent link="fuselage" />
		<child link="thrust_nozzle_left" />
		<axis xyz="0 1 0"/>
		<limit lower="-3.14/6" upper="3.14/6"/>
	</joint>

	<joint 
		name="nozzle_right"
		type="revolute">
		<origin xyz="-0.3 -0.12 0" rpy="0 0 0"/>
		<parent link="fuselage" />
		<child link="thrust_nozzle_right" />
		<axis xyz="0 1 0"/>
		<limit lower="-3.14/6" upper="3.14/6"/>
	</joint>

	<force_element name="thrust_left">
		<thrust lower_limit="0" upper_limit="36" scale_factor_thrust="1.0">
			<parent link="thrust_nozzle_left" />
			<!--<parent link="thrust_nozzle_right" />-->
			<origin xyz="0 0 0"/>
			<axis xyz="1 0 0"/>
		</thrust>
	</force_element>

	<force_element name="thrust_right">
		<thrust lower_limit="0" upper_limit="36" scale_factor_thrust="1.0">
			<parent link="thrust_nozzle_right" />
			<origin xyz="0 0 0"/>
			<axis xyz="1 0 0"/>
		</thrust>
	</force_element>
	<!--
	<transmission name="rotate_left_thrust" type="SimpleTransmission" >
		<joint name="nozzle_left" />
		<actuator name="nozzle_left_servo"/>
		<mechanicalReduction>1</mechanicalReduction>
	</transmission>
	<transmission name="rotate_right_thrust" type="SimpleTransmission" >
		<joint name="nozzle_right" />
		<actuator name="nozzle_right_servo"/>
		<mechanicalReduction>1</mechanicalReduction>
	</transmission>
	-->
</robot>

```

hongkai-dai, 2016-01-13:
When you do `reshape(dJ'*axis_world,nv,nv)`, can you check the dimension of `dJ'*axis_world`? From the left handside of line 46, `RigidBodyThrust->computeSpatialForce`, it should be reshaped to nv x nq. You may try `reshape(dJ'*axis_world,nv,nq)` in line 46, `RigidBodyThrust->computeSpatialForce`

xuhao1, 2016-01-13:
dJ'*axis_world is a 8*8 mat,and if I use reshape(dJ'*axis_world,nv,nq) in line 46, it will fail at the plus operator(nv=8,nq=9)

hongkai-dai, 2016-01-13:
I have a fix in https://github.com/RussTedrake/drake/pull/28, @xuhao1 could you try it? I tested with your code, and also the gradient of `plant.dynamics` against numerical gradient

xuhao1, 2016-01-13:
it seems work or thrust now.
I will do more tests on that to make sure.
Code still fails on other force element like Rigidbodywing and rigidbodypropellor... maybe after lunch I can fix these in the same way.

By the way, I have an additional question does drake-cpp code rely on MATLAB? I want use it on arm-linux platform like https://developer.dji.com/manifold/

RussTedrake, 2016-01-13:
No.  drake-cpp does not rely on matlab.

xuhao1, 2016-01-14:
Cool..have anyone tried it on arm?

commit 560a6564 (2016-01-13): resolves #1600
commit e995c232 (2016-01-13): Merge pull request #1601 from RussTedrake/fix_thrust_for_quats

resolves #1600