RobotLocomotion/drake #1602: Visualizer bugs when display quaternion-based model.

xuhao1, 2016-01-13:
Model on some pose,for example   0.641,0,-0.710,0(w x y z) to 0.641,0,-0.705,0 will cause display jump 
[image]
[image]

and program gives follow warning on that point
```
infeasible_constraint_name = 

    'base quat norm = 1'

Warning:   13 nonlinear infeasibilities minimized
 
> In NonlinearProgram/mapSolverInfo (line 1551)
  In NonlinearProgram/snopt (line 1427)
  In NonlinearProgram/solve (line 980)
  In Visualizer/inspector/update_display (line 306) 
```
And there are serval points can cause that problem.

When playback, it looks like my drone transmit from one pose to another pose....But as we know,SO3 is a differential manifold....
I guess when treat quaternion as a constrain here, the solver will fail at solving that constrain.Maybe normalize it direct will be better.

RussTedrake, 2016-01-13:
I don’t consider that a visualizer bug.  You are passing an ill-defined coordinate (since the norm does not equal 1).

xuhao1, 2016-01-13:
Ithat value isn't generate by my self,but by robot manipulator dynamics make it don't normalized .and also when using inspector, the norm haven't be keep either...

RussTedrake, 2016-01-13:
I looks like it came out of an optimization?  the infeasible constraints warning means that the optimization failed.  we’ll need a test case to debug it.

xuhao1, 2016-01-14:
Sure ... It looks like quaternion was solved as a constrain.

xuhao1, 2016-01-14:
Alright...my fault,when I use the up-to-date code everything goes prefect...

closed 2016-01-23 by RussTedrake