RobotLocomotion/drake #1604: Error building from master in swig_matlab step

crogers3, 2016-01-13:
On both my personal fork and a fresh distro, I get the following error when building drake master:

```
[ 65%] Performing forceconfigure step for 'swig_matlab'
Force configure of swig_matlab
[ 65%] Performing configure step for 'swig_matlab'
CMake Warning (dev) at CMakeLists.txt:17 (add_custom_target):
  Policy CMP0037 is not set: Target names should not be reserved and should
  match a validity pattern.  Run "cmake --help-policy CMP0037" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  The target name "install" is reserved or not valid for certain CMake
  features, such as generator expressions, and may result in undefined
  behavior.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/chris/drake-distro/externals/swig_matlab/pod-build
[ 66%] Performing build step for 'swig_matlab'
[ 12%] Performing patch step for 'swig-matlab'
+ test -d Tools/config
+ aclocal -I Tools/config
/home/chris/drake-distro/externals/swig_matlab/swig-matlab/autogen.sh: 11: /home/chris/drake-distro/externals/swig_matlab/swig-matlab/autogen.sh: aclocal: not found
make[7]: *** [swig-matlab-prefix/src/swig-matlab-stamp/swig-matlab-patch] Error 127
make[6]: *** [CMakeFiles/swig-matlab.dir/all] Error 2
make[5]: *** [CMakeFiles/install.dir/rule] Error 2
make[4]: *** [install] Error 2
make[3]: *** [swig_matlab-prefix/src/swig_matlab-stamp/swig_matlab-build] Error 2
make[2]: *** [CMakeFiles/swig_matlab.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

I have re-run `make options`, `make download-all` and `install_prereqs`, but I still face the same problem. The only changes I make in `make options` from default is:

- gurobi ON
- meshconverters ON
- snopt ON
- snopt_precompiled OFF

RussTedrake, 2016-01-13:
I suspect you can get unstuck by turning the swig matlab option off in make options.

crogers3, 2016-01-13:
That seems to have solved that problem, but now I get a different error further down the build:

```
[ 74%] Building CXX object systems/plants/CMakeFiles/drakeRBSystem.dir/RigidBodySystem.cpp.o
In file included from /usr/include/c++/4.8/unordered_map:47:0,
                 from /home/chris/drake-distro/drake/systems/plants/RigidBodyTree.h:7,
                 from /home/chris/drake-distro/drake/systems/plants/RigidBodySystem.h:6,
                 from /home/chris/drake-distro/drake/systems/plants/RigidBodySystem.cpp:3:
/usr/include/c++/4.8/bits/hashtable.h: In instantiation of âstd::_Hashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::__node_type* std::_Hashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::_M_allocate_node(_Args&& ...) [with _Args = {const std::piecewise_construct_t&, std::tuple<const RigidBody*&&>, std::tuple<>}; _Key = const RigidBody*; _Value = std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >; _Alloc = Eigen::aligned_allocator<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> > >; _ExtractKey = std::__detail::_Select1st; _Equal = std::equal_to<const RigidBody*>; _H1 = std::hash<const RigidBody*>; _H2 = std::__detail::_Mod_range_hashing; _Hash = std::__detail::_Default_ranged_hash; _RehashPolicy = std::__detail::_Prime_rehash_policy; _Traits = std::__detail::_Hashtable_traits<false, false, true>; std::_Hashtable<_Key, _Value, _Alloc, _ExtractKey, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits>::__node_type = std::__detail::_Hash_node<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >, false>]â:
/usr/include/c++/4.8/bits/hashtable_policy.h:519:8:   required from âstd::__detail::_Map_base<_Key, _Pair, _Alloc, std::__detail::_Select1st, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits, true>::mapped_type& std::__detail::_Map_base<_Key, _Pair, _Alloc, std::__detail::_Select1st, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits, true>::operator[](std::__detail::_Map_base<_Key, _Pair, _Alloc, std::__detail::_Select1st, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits, true>::key_type&&) [with _Key = const RigidBody*; _Pair = std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >; _Alloc = Eigen::aligned_allocator<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> > >; _Equal = std::equal_to<const RigidBody*>; _H1 = std::hash<const RigidBody*>; _H2 = std::__detail::_Mod_range_hashing; _Hash = std::__detail::_Default_ranged_hash; _RehashPolicy = std::__detail::_Prime_rehash_policy; _Traits = std::__detail::_Hashtable_traits<false, false, true>; std::__detail::_Map_base<_Key, _Pair, _Alloc, std::__detail::_Select1st, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits, true>::mapped_type = Eigen::Matrix<double, 6, 1>; std::__detail::_Map_base<_Key, _Pair, _Alloc, std::__detail::_Select1st, _Equal, _H1, _H2, _Hash, _RehashPolicy, _Traits, true>::key_type = const RigidBody*]â
/usr/include/c++/4.8/bits/unordered_map.h:600:20:   required from âstd::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type& std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::operator[](std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::key_type&&) [with _Key = const RigidBody*; _Tp = Eigen::Matrix<double, 6, 1>; _Hash = std::hash<const RigidBody*>; _Pred = std::equal_to<const RigidBody*>; _Alloc = Eigen::aligned_allocator<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> > >; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::mapped_type = Eigen::Matrix<double, 6, 1>; std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::key_type = const RigidBody*]â
/home/chris/drake-distro/drake/systems/plants/RigidBodySystem.cpp:49:52:   required from here
/usr/include/c++/4.8/bits/hashtable.h:727:6: error: no matching function for call to âEigen::aligned_allocator<std::__detail::_Hash_node<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >, false> >::construct(std::_Hashtable<const RigidBody*, std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >, Eigen::aligned_allocator<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> > >, std::__detail::_Select1st, std::equal_to<const RigidBody*>, std::hash<const RigidBody*>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::__node_type*&, const std::piecewise_construct_t&, std::tuple<const RigidBody*&&>, std::tuple<>)â
      _M_node_allocator().construct(__n, std::forward<_Args>(__args)...);
      ^
/usr/include/c++/4.8/bits/hashtable.h:727:6: note: candidate is:
In file included from /home/chris/drake-distro/build/include/eigen3/Eigen/Core:256:0,
                 from /home/chris/drake-distro/drake/systems/System.h:5,
                 from /home/chris/drake-distro/drake/systems/plants/RigidBodySystem.h:4,
                 from /home/chris/drake-distro/drake/systems/plants/RigidBodySystem.cpp:3:
/home/chris/drake-distro/build/include/eigen3/Eigen/src/Core/util/Memory.h:729:10: note: void Eigen::aligned_allocator<T>::construct(Eigen::aligned_allocator<T>::pointer, const T&) [with T = std::__detail::_Hash_node<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >, false>; Eigen::aligned_allocator<T>::pointer = std::__detail::_Hash_node<std::pair<const RigidBody* const, Eigen::Matrix<double, 6, 1> >, false>*]
     void construct( pointer p, const T& value )
          ^
[... cut ...]    candidate expects 2 arguments, 4 provided
make[6]: *** [systems/plants/CMakeFiles/drakeRBSystem.dir/RigidBodySystem.cpp.o] Error 1
make[5]: *** [systems/plants/CMakeFiles/drakeRBSystem.dir/all] Error 2
make[4]: *** [all] Error 2
make[3]: *** [drake-prefix/src/drake-stamp/drake-build] Error 2
make[2]: *** [CMakeFiles/drake.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [all] Error 2
```

This is very reminiscent of the error that I've been seeing from last week. Is there a fix for this?

RussTedrake, 2016-01-13:
Sorry the fix is in my prius pull request, waiting to merge...

tkoolen, 2016-01-14:
Another workaround is to upgrade g++ (don't do this if you're not comfortable with it of course, but wanted to add it for posterity).
install g++-4.9:
```
sudo apt-get install g++-4.9
```
register 4.8 and 4.9 with update-alternatives:
```
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-4.8 40
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-4.9 50
```
you can switch between 4.8 and 4.9 using
```
sudo update-alternatives --config g++
```

rdeits, 2016-01-19:
Looks like you found a missing prereq on ubuntu. Thanks for letting us know! Fix is incoming...

PR #1626 (2016-01-20): update swig_matlab to get more prereqs on ubuntu
fixes #1604
commit a715b444 (2016-01-20): update swig_matlab to get more prereqs on ubuntu

fixes #1604
closed 2016-01-20 by RussTedrake