RobotLocomotion/drake #1607: Visuallizer failed after switch to source install.

xuhao1, 2016-01-14:
When I try use up-to-date source, after compile the code
```
>> run /Users/xuhao/Software/drake-distro/drake/addpath_drake.m
Adding /Users/xuhao/Software/drake-distro/drake/../build/matlab to the matlab path

Warning: Name is nonexistent or not a directory: /Users/xuhao/Software/drake-distro/drake/pod-build/bindings/matlab 
> In path (line 109)
  In addpath (line 88)
  In addpath_drake (line 64)
  In run (line 96) 
```
and when call the visuallizer
```
>> fucktest
Warning: This model has no mex pointer. Only checking collisions between terrain contact points and terrain 
> In WarningManager/warnOnce (line 21)
  In RigidBodyManipulator/collisionDetect (line 101)
  In RigidBodyManipulator/compile (line 860)
  In RigidBodyManipulator/addRobotFromURDF (line 60)
  In RigidBodyManipulator (line 75)
  In fucktest (line 2) 
attempting to launch the drake director
Error using BotVisualizer (line 48)
Failed to automatically start up a viewer (or to receive the ack, see
https://github.com/RobotLocomotion/drake/issues/317)
```

belong this, program is OK when using pre-built distribution and when I open the visualizer in matlab with the pre-built code and don't close it and switch to source-built distribution the program is OK too....
In drake-distro/build/bin folder
```
./drake-visualizer
```
can launch the visualizer..

xuhao1, 2016-01-14:
And use matlab to open the ddConsoleApp failed
```
>> systemWCMakeEnv([fullfile(pods_get_bin_path,'ddConsoleApp'),' -m ddapp.drakevisualizer &> ddConsoleApp.out &'])

ans =

     1
```

RussTedrake, 2016-01-17:
you'll have to show us the output for us to debug.

RussTedrake, 2016-01-23:
the app name did change around this time.  perhaps you caught us between pull requests.  will close this unless we can get more details.

closed 2016-01-23 by RussTedrake